Klipper3d/klipper #1232: 4 point mesh leveling

thucar, 2019-02-11:
I'm almost finished with my 3D printer upgrade. As part of the upgrade, I'm moving from Marlin to Klipper. 

I love the flexibility so far. One thing I have not managed to figure out yet though, is how to set up a bed leveling system where I would probe 4 fixed positions (corners) of the bed. The reason for this is that I have aluminium foil attached to the 4 corners of my glass print bed. 

I've tried to set the bed leveling conf from 3,3 to 2,2 points but that throws up an error. Looking at the code for the underlying function (can't remember the name of it off the top of my head) requires 3 points as a minimum. Would it be ok to go in and change that minimum to 2 instead? 

Or am I approaching this the wrong way?

Arksine, 2019-02-11:
No,  bed_mesh requires a minimum of 3 samples along each axis to generate a mesh.  You may want to look into bed_tilt, although my understanding is that it works better with 3 points than 4.

thucar, 2019-02-12:
Perfect! I knew I was missing something. 3 point tilt calibration is working like a charm!

closed 2019-02-12 by thucar