Klipper3d/klipper #1234: Move exceeds maximum extrusion (1084292.768mm^2 vs 0.640mm^2)

ArdentSquirrel, 2019-02-11:
I keep getting shut down with a message "Move exceeds maximum extrusion (1084292.768mm^2 vs 0.640mm^2)".  I do not see anything in the gcode that is causing overextrusion, and certainly not by 1 million mm^2.

I am running a Piper2 CoreXY with multiple MCUs, with 4 z motors.  My feeling is that this might be related to mesh bed levelling.

It fails at this line in the gcode:

M105
M109 S185
M104 T0 S175
G1 F1200 E9.5
;MESH:sub_millimeter_calibration0.stl
G0 F7200 X77.375 Y117.4 Z1.2
M104 S195

[klippy.maxextrusion1.log](https://github.com/KevinOConnor/klipper/files/2851925/klippy.maxextrusion1.log)

[CFFFP_sub_millimeter_calibration0.gcode.txt](https://github.com/KevinOConnor/klipper/files/2851932/CFFFP_sub_millimeter_calibration0.gcode.txt)

Also fails in the same spot on another gcode:
M105
M109 S190
M104 T1 S195
G1 F1200 E9.5
;MESH:Checkerboard_2-color_Kobayashi_Solid_HingesColor1.stl
G0 F7200 X165.451 Y59.77 Z1.2

[CFFFP_Checkerboard_2-color_Kobayashi_Solid_HingesColor1.gcode.txt](https://github.com/KevinOConnor/klipper/files/2851935/CFFFP_Checkerboard_2-color_Kobayashi_Solid_HingesColor1.gcode.txt)
[klippy.maxextrusion2.log](https://github.com/KevinOConnor/klipper/files/2851938/klippy.maxextrusion2.log)

Arksine, 2019-02-11:
Its possible that mesh leveling is responsible for this.  It seems to be related to the toolchange after fade begins.  I'll see if I can track it down.

In the meantime, you can try two things to see if its fade related:
1) Set the fade_target option of bed_mesh to 0, ie:
```
fade_target: 0
```
2) Alternatively you can try it with fade disabled by setting the fade_end option of bed_mesh to 0:
```
fade_end: 0
```

BlackStump, 2019-02-11:
it would be your purge line
G1 F1200 E9.5
the error message does give the clue

I must admit it would be nice if a temporary disable command to allow a purge then enable again after the purge was available
https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L105

Arksine, 2019-02-11:
@BlackStump That purge should be able to pass through since it isn't a kinematic move.  Klipper thinks its getting a small move, specifically a move with the length of .000021073808 mm if my math is correct.

@ArdentSquirrel I think I have narrowed down the source of the issue.   It is related to fade, if you disable it you should be able to print.   I'll work on a fix and get it submitted as soon as I am able.

ArdentSquirrel, 2019-02-11:
@Arksine I'll change the fade and try again.  Thanks for the quick reply!

mattkenn4545, 2019-02-12:
Could this have anything to do with poor nozzel height consistency across the bed? Ie even though I'm fairly sure my bl touch is correctly setup prints at the rear of the bed have the nozzel far to close to the bed, the middle of the bed is perfect and the front of the bed is too high. I've tried using multiple samples, 3x3, 5x5, 11x11 it's all the same symptom. Note that the z rod does move slightly ie it is using the mesh and its not saved to config. I get a fresh mesh every print. My bed is also fairly well trammed and the 0.2mm variation across the surface.

Arksine, 2019-02-12:
@mattkenn4545  No.  It is very specific bug related to the purge after a tool change.  

First layer problems can almost always be put in one of the following buckets:
1) bed_mesh is not configured correctly
2) inaccurate probe
3) Mechanical issues with the printer

ArdentSquirrel, 2019-02-27:
I set fade_target: 0, and still got the max extrusion error.  Going to try again with fade_end: 0.  This is only happening during dual extrusion (because of the purge line).

[klippy_20190227.log](https://github.com/KevinOConnor/klipper/files/2911765/klippy_20190227.log)

Arksine, 2019-02-27:
@ArdentSquirrel You need to pull a recent version of Klipper to get the fix for the purge issue (your log is showing an older version).  It should work with fade enabled.

ArdentSquirrel, 2019-02-27:
@Arksine I just noticed that advice on 1252 . :-)  Pulling now.

ArdentSquirrel, 2019-02-28:
I pulled the latest version, re-enabled fade, and everything is working.  Thank you!

closed 2019-02-28 by ArdentSquirrel