Klipper3d/klipper #1237: MCU 'mcu' shutdown: Move queue empty

popshansen, 2019-02-11:
Sometimes I get this error even it is printing from virtualsd on the Beaglebone Black. It is only running klipper and octoprint. I have stopped using webcam on the Beaglebone as it causes problems with G29. So the CPU load should be okay when queue is empty.
It is the last print in the klippy.log where the issue happens.
[klippy.log](https://github.com/KevinOConnor/klipper/files/2853238/klippy.log)

KevinOConnor, 2019-02-14:
I think the PRU may have gotten temporarily overloaded.  That's certainly surprising.

What type of stepper drivers do you have?  If you are not using DRV8825s, you could try setting a custom step delay of 1us in "make menuconfig" - just to see if that helps.  (At the time of the failure, the Y axis was moving at around 100,000 steps per second and I wonder if the step delays are contributing to an issue in the PRU.)

-Kevin

KevinOConnor, 2019-02-14:
FYI, you will need to upgrade to the latest code to apply a custom step delay.

-Kevin

popshansen, 2019-02-14:
The steppers is Raps128, where X and Y is running at 128 microsteps, and I think I reduced the Z and E to 64 microsteps, in order to avoid some issues in Klipper. I also have HOMING_STEP_DELAY=0.000005 or I get timing issues.

I'll do the upgrade when I get home from work, and se how it works.

popshansen, 2019-02-14:
What are you looking at in the log, in order to see steps per second ?

popshansen, 2019-02-14:
Now I have tested the "custom step delay of 1us", and that result in a new "MCU 'mcu' shutdown: Move queue empty". I have attached this klippy.log
[klippy.log](https://github.com/KevinOConnor/klipper/files/2866761/klippy.log)

KevinOConnor, 2019-03-04:
Okay, I've added this to my list of things to look at.  Not sure when I will get to it, however.

-Kevin

popshansen, 2019-03-04:
If I do FIRMWARE_RESTART between prints, I don’t encounter the issue that often, and it is with the same printing speed (60ms). I don’t know if it help investigating the issue.

KevinOConnor, 2019-03-31:
I've found a problem in the PRU timer handling (commit 72d0c770).  Both of your errors resulted immediately after a 32-bit timer rollover, so there's a very good chance this is the root cause of your issues.  I recommend you upgrade to the latest code.  Let us know if you continue to see issues.

-Kevin

KevinOConnor, 2019-04-14:
I'm guessing this issue is fixed now - I can reopen if the problem occurs again.

-Kevin

closed 2019-04-14 by KevinOConnor

popshansen, 2019-04-16:
Hi Kevin,

Yes, it seems to have fixed this issue. I have done several prints without have to FIRMWARE_RESTART between prints.

Thanks.

KevinOConnor, 2019-04-16:
Great!  Thanks for the follow up.

-Kevin