Klipper3d/klipper #1244: Homing anomaly

thucar, 2019-02-12:
I'm using sensorless homing on X and Y with my TMC2130 drivers. Since my print head moves off the bed area then I've defined negative min_positions as well as negative position_endstop values for both of those axes.

```
[stepper_x]
step_pin: ar54
dir_pin: ar55
enable_pin: !ar38
step_distance: .0125
#endstop_pin: ^ar3
endstop_pin: tmc2130_stepper_x:virtual_endstop
position_min: -60
position_endstop: -60
position_max: 190
homing_speed: 50

[stepper_y]
step_pin: ar60
dir_pin: !ar61
enable_pin: !ar56
step_distance: .0125
#endstop_pin: ^ar14
endstop_pin: tmc2130_stepper_y:virtual_endstop
position_min: -50
position_endstop: -50
position_max: 160
homing_speed: 60
```

In addition, when probing my first z-height position, my print head is also off the bed. The exact position is X10,Y50 if considering endstop positions 0;0 However, since I've entered negative endstop positions, then 0;0 refers to my actual bed 0;0 - exactly as I would expect

However, when creating homing_override, I ran into issues. At first I tried to have the print head move to 
-10;-50 for probing. That did not work because apparently the printer added the two negative values from position_endstop and G0 moves together and tried to move to -70; -100 or some such.

Then I used positive values which got me the result I needed. 
```
[homing_override]
set_position_z: 0
gcode:
	G0 F1500
	G0 Z5
	G28 X Y
	G0 X10 Y50
	G28 Z   # triggering 2 times
	G0 Z10
```
However - when I would send a new G28 command after moving the axes around, doing bed tilt calibrations, etc. It then tries to Z-probe in the actual X10;Y50 position on my print bed - thus driving the nozzle and probe into my glass bed.

KevinOConnor, 2019-02-14:
In order for us to assist, you'll need to attach the Klipper log file from the event - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

thucar, 2019-02-14:
I've narrowed the issue down to Octoprint. 
First, I changed my z-homing to the other corner of my print bed, so that the position does not contain negative numbers. Then I excluded X and Y axis from the homing override and only use it for Z.

This is my override now:
```
[homing_override]
set_position_z: 0
axes: z
gcode:
	G0 F1500
	G0 Z5
	G0 X148 Y0
	G28 Z   # triggering 2 times
	G0 Z10
```
Now when I issue G28 Z command after homing X and Y, it works flawlessly every time. But when I hit the Z-homing button in Octoprint, it runs only the X148 move, and not Y0. I have a feeling it's Octoprint thinking that homed Y==Y0

So I just do not use Octoprints Z-Homing button anymore

closed 2019-02-22 by thucar