Klipper3d/klipper #1246: Unknown command:"PROBE_CALIBRATE"

teikjoon, 2019-02-12:
Hello,

Am trying to get my EZABL working...I managed to get it installed and the offsets set properly.

But when I try to follow the instructions over at https://github.com/KevinOConnor/klipper/blob/work-bed-level-20190206/docs/Probe_Calibrate.md I hit a snag...I get **Unknown command:"PROBE_CALIBRATE"**

Have used git pull to fetch the latest source and restarted the klipper service...still get this.

Hywelmartin, 2019-02-12:
are your on the right Branch??? 
work-bed-level-20190206

KevinOConnor, 2019-02-14:
The code wasn't merged until earlier today.  If you're still having an issue, you'll need to attach the klipper log file from the event - see:
https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

teikjoon, 2019-02-15:
@KevinOConnor I updated klipper and it works, thanks!

closed 2019-02-15 by teikjoon