Klipper3d/klipper #1247: Changing extruder steps/mm mid-print

thucar, 2019-02-12:
I'm running through the calibration steps with my new printer and I was wondering how does everyone handle Extruder calibration with Klipper? I've done the filament measurement and raw calculation of steps from there. However, this has never gotten me solid results. I would usually run a print with high infill afterwards and just tune in the exact value looking at the infill and top shell surfaces.

With Klipper I have not found a way to input new steps values in mid-print.

Hywelmartin, 2019-02-12:
e-steps isn't run-time configurable..

On the other hand flow is.. so altering the flow during print will render the same function.

thucar, 2019-02-14:
Thanks! That's what I ended up using. I feel dumb, it did not occur to me before.

closed 2019-02-14 by thucar