Klipper3d/klipper #1248: Can not work with Bed Visualizer plugin

YaronSoffer, 2019-02-12:
Hi There,

This is more a feature request than an actual bug/ problem. There seem to be some differences in how Klipper is outputting his string regarding mesh bed leveling and how it is done in Marlin. 
As a result, the plugin fails in reading the mesh bed level. Can you fit the string format, so it will work?
I have to say that I am using your firmware for the last few months. I have two machines one is with the Duet board and one running Klipper with Ramps and TMC2130, Lcd, Induction sensor. Klipper is a way better than Dute board. Do you have a Patron page?

ArDrift, 2019-02-12:
The plugin can indeed read the values from klipper, if the method you're using to show the output is correct: https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/92#issuecomment-430783036
You should set this
```
@BEDLEVELVISUALIZER
BED_MESH_OUTPUT
```
in the plugin settings and it will work.

YaronSoffer, 2019-02-13:
working w/o a problem. Thanks

KevinOConnor, 2019-02-14:
Looks like you've found the solution.

> Do you have a Patron page?

Thanks.  There is: https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#how-can-i-donate-to-the-project

KevinOConnor, 2019-02-25:
I'm going to close this as it looks like a solution was found for the original issue.

-Kevin

closed 2019-02-25 by KevinOConnor