Klipper3d/klipper #1249: Feature Request: Mini IR Probe Analog Mode and Slowdown when probing

Armadillo1, 2019-02-12:
Hi!

I am in the process of switching from Duet to Duet+Klipper on my Voron 2.1. I'm using an IR probe by dc42 (https://miscsolutions.wordpress.com/mini-height-sensor-board/). Duet has a way to force the probe into an analog mode which allows the probe to have two trigger heights - one slightly higher (probe value <~470 and one lower (value>=500). The benefit is that the printhead slows down just before the lower point, allowing for higher probe speed, while keeping accuracy due to lower speed. Any plans to implement that ?

P.S. I'm not much into python, but am willing to give it a go if noone else is interested in this.
Thanks!

nythil, 2019-02-13:
Hi! I too have the IR height probe by dc42. I've been studying it recently in order to fix my probing issues (which have turned out to be EMI noise). The probe does in fact have an analog mode, in which it can additionally signal a) close proximity to trigger point and b) phototransistor saturation, which basically means that the triggering was unreliable. Both of these could be useful to Klipper. I think the greatest difficulty would be the overhead of continuously sampling the probe via ADC, especially on ATmega MCUs.

I'd like to see this feature added one day, however a much simpler solution would be good enough for now. Klipper could use the addition of making probing work the same way X/Y/Z endstops do, i.e. first approaching the endstop fast until triggered, then pulling back a bit and making a slow, precise approach. Once I resolve the noise issues in my 3D printer (which is likely going to take a couple of days), I'll happily have a go at implementing this. Unless someone else does it sonner :)

Armadillo1, 2019-02-13:
I managed to successfully switch to Klipper and get it printing. I'm leveling with the IR probe'a in digital mode at a speed of 5 and it's working fine with pretty decent repeatability on black matte printbite. But the slowdown would make things faster. 
Haven't thought about issues on 8bit boards. I suppose duet should handle that. I might try to give it a go in a few weeks time unless someone's more keen. I never tried python so it might take way more than a few weeks though😁

KevinOConnor, 2019-02-14:
It's fine if someone wants to work on this.  I suspect it would be a lot of work to trigger the endstop halt on an analog input.  The changes would need to be done in the micro-controller C code (the motion halting during homing/probing is implemented in the micro-controller to ensure a timely response).

-Kevin

KevinOConnor, 2019-03-04:
I'm going to close this as I think the original discussion has wrapped up.  Feel free to submit a PR with the support when ready.

-Kevin

closed 2019-03-04 by KevinOConnor