Klipper3d/klipper #1250: Printer wont home y

jamesrooke54, 2019-02-13:
hi all

so the issue I'm having is octoprint connects and detect temps fine but when I run a g28 command the x homes fine but the bed just moves forwards until the end of its axis (opposite end to y stop) and then I get the error

Send: G28
Recv: !! Failed to home y: Timeout during endstop homing
Recv: ok

the printer is a cr-10s and the log is attached along with the config file, any help is appreciated, I'm fairly new to this and very new to Klipper but ill keep up where I can

[printer config.txt](https://github.com/KevinOConnor/klipper/files/2861025/printer.config.txt)

[klippy.log](https://github.com/KevinOConnor/klipper/files/2861017/klippy.log)

joonadec, 2019-02-13:
Bed moves the opposite direction? Try putting an exclamation mark (!) in front of the dir_pin in the [stepper_y] section to reverse the motor.

You could also double check the endstops to see if they're triggering by issuing M119

KevinOConnor, 2019-02-14:
Have you completed the config checks described at: https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md ?

jamesrooke54, 2019-02-14:
hi so ive completed the config checks and all seems to be ok i can buzz all of the steppers and they work fine i have tried the ! in front of the pin and if i move the bed to trigger the endstop then it tries to move the wrong way and if i move the bed to the middle and run home it does the same thing just moves away from the trigger slowly until it reaches the opposite end of the frame

i almost seems to be an issue with the G28 command rather than the machine it just seems to move forward forever or moves in the wrong direction

jamesrooke54, 2019-02-14:
ok so I've just reloaded a default of the cr-10s cfg file and its now working and homing fine

I have no idea what I must have changed but clearly, I did something thanks for the support and replies guys

closed 2019-02-14 by jamesrooke54