Klipper3d/klipper #1252: Tool change problem

nandinho84, 2019-02-14:
I'm installing a dual extrusion system with shared nozzle. However I can't change the tool properly. 
The first tool change is completed successfully, however, the next tool changes does not behave like it should - only retract the filament - it doesn't extract the new one to the nozzle.

My tool change script on S3D(working ok with Marlin firmware):
T[old_tool]
G92E0
G1 E-10 F4500 ; retract the filament from melting zone
G92E0
G1 X30 Y315 F3000 ;move the hotend to bed edges or corners to finish the filament swap job. normally over the prim pillar is ideal
G4 S1 ;wait for 1 second to reduce the filament stringing
G1 E-115 F1200 ; retract the filament cross the intersection back to the it?s own tube waiting for the Gcode summon. distance is depending on your hardware setup
G92E0

T[new_tool]
G92 E0
G1 E125 F800 ;extrude the new selected filament to the nozzle 
G92 E0

At first, the firmware showed me an error due to the max_extrude_cross_section (initially I've not set this parameter), then I put a value that wouldn't trigger the error, but the problem persisted (the same behaviour as with max_extrude_cross_section disabled.

Here is my log file:

[klippy.zip](https://github.com/KevinOConnor/klipper/files/2862950/klippy.zip)

Arksine, 2019-02-14:
Its possible that this is a duplicate of #1234.  Try setting fade_end in the [bed_mesh] section to disable it to see if it solves your problem.

I'm currently testing a fix for this and it shouldn't be long before its submitted.

KevinOConnor, 2019-02-26:
Several fixes were recently added.  Can you retry with the latest version of Klipper?

-Kevin

nandinho84, 2019-03-10:
@KevinOConnor  Sorry for the late reply, but I didn't had my dual extrusion setup ready to give it a try before today.
I've retried again today, but the same problem remains - The first tool change is completed successfully, however, the next tool changes does not behave like it should - only retract the filament - it doesn't extract the new one to the nozzle. The same pattern is repeated continuously (only retraction during tool change). This only happens with dual extrusion prints.
I must tell you that the same gcode is working as it should with marlin.
I'm really enjoying Klipper. However, this is not working with my single nozzle dual extruder setup.
Here is my log file:
[klippy (2).zip](https://github.com/KevinOConnor/klipper/files/2949014/klippy.2.zip)

Arksine, 2019-03-10:
@nandinho84 I'm looking at your log and I can't match the hash of your git version to any commits in Keivn's repo.  Are you sure you are on the master branch?

Regardless, the hash matches the build you flashed to the MCU, so I don't believe you have pulled the latest commits which would have the fix.

nandinho84, 2019-03-10:
About 10 days ago I've made a git pull... I think I'm on the master branch but tomorrow I'll check it out. Are you saying that I must reflash the MCU's? That I know I haven't done! If that's the case, tomorrow I'll update the firmware and the MCU's again and try again

Arksine, 2019-03-10:
You shouldnt need to reflash the MCU.   The log reports the git hash for the host,  as well as the hash and build date for the MCU.   I use this information to see if you are running a version that contains the recent patch for bed_mesh that should fix this issue.

I couldn't find the hash using the search function, which tells me that you are either on another branch or on a fork that isn't pushed to github.  However since both the host and mcu share the same hash, I can assume that your host is running a host version that was last updated on or before the MCU build date, 2-6-19.

nandinho84, 2019-03-11:
Sorry! I wasn't in the master branch and I didn't notice. Everything is working ok so far after the update.

closed 2019-03-11 by nandinho84