Klipper3d/klipper #1255: Unit tests

drola, 2019-02-14:
I quickly glanced through the source code of LinuxCNC, Marlin Firmware (v2), RepRap Firmware, Smoothieware, and Klipper.
As a software engineer, I'm worried about the lack of tests. Klipper's firmware seems to be at a good place to write thorough tests for. Hardware layer is conveniently abstracted. Thus it is possible to mock hardware for testing.

Do you see this as a problem too? How open are you for pull requests in this direction?

trevjonez, 2019-02-14:
I am +100 on this. I also like how @drola wrote it as a user story. (unintentionally?)

I started playing with pytest and have a worked up a docker image I can use locally with pycharm to run tests without doing a full install of klipper on my desktop machine.

As far as useful tests go, outside of some utility functions I found it tricky as there is a bit of coupling to work with. In my mind there will need to be _some_ refactoring to extract the most value from the testing efforts.

drola, 2019-02-15:
The intention behind this request is to proceed with #258 (and #256). 

The first step would be to write tests for existing code around `move_transform`. After that, another transform could be added: XYSkew. And finally, to bring all this together, `klippy/gcode.py` needs to be changed to support multiple transforms.

KevinOConnor, 2019-02-15:
I'm open to increasing the test case coverage.  It would probably be a good idea to go over the high-level framework you have in mind.

FWIW, I generally prefer "blackbox" testing to unit tests.  (In my experience, blackbox tests are more work to build, but have lower ongoing maintenance.)

Be sure to check out the existing test cases in the test/ directory.  We are currently running tests on all the printer config files and a good number of the g-codes are also run.  Though, the micro-controller is only compile tested, and the host tests just verifies that no internal errors are raised.  The original intent of the src/simulator/ code was to support blackbox testing of the micro-controller, but that effort was never completed.

-Kevin

drola, 2019-02-15:
Proceeding with the simulator is a good idea too. If I understand correctly, the intent is to have a simulated microcontroller that would dump received commands and simulate signals from homing switches, temperature probes and other input?

KevinOConnor, 2019-03-04:
I'm going to close this as I think the original discussion has concluded.  Feel free to submit a PR with additional tests.  Or, I can reopen this if there is additional questions.

-Kevin

closed 2019-03-04 by KevinOConnor