Klipper3d/klipper #1256: Multi MCU slave master power sequence.

wizhippo, 2019-02-14:
Do all mcu's have to come up at the same time?

Does the master have to have xy control?

My use case is that I wish to have one master MCU come up first.  The master mcu would have a few temperature fans that monitor the mosfet temps and shutdown if temps are surpassed. That MCU controls the power on the HV psu which also powers the second mcu.  This order is important as my second MCU cannot be powered by usb first and then have the HV psu turned on as that can blow the tmc drivers.

Possible?

Hywelmartin, 2019-02-15:
https://www.watterott.com/en/SilentStepStick-Protector would be beneficial and this one even protect the board if you move the steppers by hand.. to fast...

trevjonez, 2019-02-15:
@Hywelmartin what effect do those step stick protectors have on tmc drivers with stallguard?

Hywelmartin, 2019-02-15:
@trevjonez  I don't use that feature.. so I don't know.. I use them only to get the right power on sequence, but that could be done with a single diode I believe..

cmock, 2019-02-16:
StallGuard (i.e. sensorless homing) works fine for me with those protector boards; if you look at the schematic, the diodes clamp the stepper motor lines to GND and VCC, they are not inline like the TLsmoothers. And the ninth diode makes sure that Vm is fed from the 5V rail, which is the relevant part here.

KevinOConnor, 2019-02-18:
The current code expects to bring up all the micro-controllers at system startup.  It's certainly possible to change the code to do something else.  I do wonder if adding code complexity to accomplish that is worthwhile.  (Might be better, for example, to add a "hibernate mode" so that klipper can power down components until the next significant g-code command.)

-Kevin

wizhippo, 2019-02-20:
@cmock Thanks for clarifying that.  I'll grab some v2 protector board for the tmc2130's

closed 2019-02-20 by wizhippo