Klipper3d/klipper #1257: CR-10 Config Won't Load

Deneteus, 2019-02-15:
I am having an issue where I am trying to load the CR-10 config (printer-creality-cr10-2017.cfg) from the config folder via 'make config' and its displaying an error in the background and not loading the config file.

[image]

jakep82, 2019-02-15:
It's very important to follow the installation instructions.  You don't load the config file in that part of the installation.

https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md#building-and-flashing-the-micro-controller

Deneteus, 2019-02-15:
You instructions should actually say that as well as explain that you have to manually save.

When I tried it the first time I couldn't get it to upload even though the settings were correct. 

[image]

> It's very important to follow the installation instructions. You don't load the config file in that part of the installation.
> 
> https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md#building-and-flashing-the-micro-controller

s1mpleman, 2019-02-15:
I found that bit pretty self explanatory tbf. It also doesn't tell you to upload a file to it.

Deneteus, 2019-02-15:
What doesn't make any sense is telling people to manually setup their settings when there should be config files for that. You don't have to do that with Marlin example configs so why would it make any sense to have to do it with Klipper? You should already have pre-made configs that you can use to setup the entire process.

NotExpectYet, 2019-02-15:
For me that makes sense only development projects with multiple developers that makes sense to me. Having pretty much-done something off your own back (not currently aware of Klipper's history, so correct me if I'm wrong here) and shared the code with the community for free seems more like "your lucky to get any configs and documentation at all" type of situation so I don't agree at all. Marlin isn't Klipper, and Klipper doesn't have the development resources Marlin seems to have. Don't get me wrong, I would like that and currently have written a full walkthrough of the installation process I did for my CR10, I've shared this on my own personal Wiki which is currently down for repairs and domain swap, otherwise I'd have shared it with the communities I'm involved in with Klipper. I don't speak for the developer here at all, but heck it's github, if you think something is confusing in the documentation then create a pull request for it? The developer seems happy to have people help with this project.

Deneteus, 2019-02-15:
Actually I already made a patch request for it with the changes that should be there. If anything else needs to be changed I have no problem doing it.

Deneteus, 2019-02-15:
BTW I used Marlin as an example because in all of the groups people are either leaving Marlin or Repetier for Klipper once they see peoples posts about it.

NotExpectYet, 2019-02-15:
> BTW I used Marlin as an example because in all of the groups people are either leaving Marlin or Repetier for Klipper once they see peoples posts about it.

It just doesn't seem necessary to me to point that out, no drama. That's grand though, it was a complete game changer for me and I'm a rather seasoned installer of random things I find on here and I had some trouble with the documentation and how it's laid out so I get how frustrating it can be. I resolved all that though by checking here, for every issue I ran into and low and behold it was in the documentation somewhere. Hopefully, then with your changes and others, we can make it more understandable because I wholeheartedly believe this is a wonderful way of managing the printers.

Deneteus, 2019-02-15:
It's pretty necessary because some people have no idea that others go through instructions thinking about how other people will interpret it. I went through it verbatim the first time and it failed so I wiped Octoprint and started over which is how we got here.  If I figured it out for myself and never posted anything about it then you wouldn't know that the instructions need to take into account for people that don't follow instructions the same way. If people see there are other options they will explore them when the instructions don't work for them. As a technical trainer the following comment is pretty dramatic.  

[image]

NotExpectYet, 2019-02-15:
This isn't a business though and he's not getting paid for it so I don't find that dramatic at all. He did what he could how he did, its as simple as that for me. I was also referring to the wide amount of GitHub pages I've seen with applications I've installed a modecom of reason is required to read them, sometimes the documentation isn't complete. Asking why isn't productive to solving the issue to me as well, who cares? If you figured it out just post your fix, it doesn't require drama. Your initial comment said "the instructions don't state you shouldn't up load a file" I would have presumed as a technical trainer you would know most documentation tells you what to do, not what not to do.

Deneteus, 2019-02-15:
1. Never said this was a business. If it it isn't the documentation should still be well thought out. 
2. Never mentioned anything about said creator. You took it as a personal affront.
3. The documentation doesn't give alot of reason behind the commands at all.

It actually doesn't even tell you to exit the config program or to save it for that matter.
It also has you stop the service when it should have been stopped before you tried to do anything else.
It also tries to make the claim that ACM0 is the most common port and that isn't true at all. I have 15 printers. The only one that comes up as ACM is a Wanhao Ultimaker Clone.
4. If you don't expect people to explore the application when instructions don't get the desired result then you aren't very skilled in technical support.
If it didn't break anything it shouldn't be a problem.
5. If I wasn't being productive I wouldn't have submitted the changes hours ago without anyone even asking about it. But ya know I am doing things without thinking about it is the assumption that you had from the start of this conversation.

FHeilmann, 2019-02-16:
There is always going to be a tradeoff here. Some people might be fine with the docs as they are, but there will always be people that want more details. In the end its up to the owner of the project how detailed he wants the instructions to be while still being lean and concise. 

Submit your suggestions for changes as a pr to this repo and they might get accepted. Otherwise you'll have to accept the fact this project does not share your view of how much detail is required. In that case it's easy enough to make these instructions available elsewhere. 

There's nothing to gain from dragging out this discussion.

FWIW I also did not have issues with the lack of detailed instructions that you perceive.

Deneteus, 2019-02-16:
> 
> There's nothing to gain from dragging out this discussion.
> 
> FWIW I also did not have issues with the lack of detailed instructions that you perceive.

Says the person dragging out the discussion. I am sure you speak for everyone as well.  ;)

pecirep, 2019-02-16:
The point still stands. If you believe that the docs need to be more detailed, you are free to submit a Pull Request that I'm sure will be accepted if it improves the quality of the instructions.

jakep82, 2019-02-16:
> > There's nothing to gain from dragging out this discussion.
> > FWIW I also did not have issues with the lack of detailed instructions that you perceive.
> 
> Says the person dragging out the discussion. I am sure you speak for everyone as well. ;)

He speaks for me.  Now I'm going to unsubscribe so I don't need to see any more petty bickering.

Deneteus, 2019-02-16:
> The point still stands. If you believe that the docs need to be more detailed, you are free to submit a Pull Request that I'm sure will be accepted if it improves the quality of the instructions.

Pull Request Submitted.

KevinOConnor, 2019-02-18:
FYI, as of commit 021264a6, the "make menuconfig" tool will no longer have Save/Load buttons.  These buttons were not useful and they appear to have caused confusion with some users.

-Kevin

closed 2019-03-04 by KevinOConnor