Klipper3d/klipper #1259: TMC2660 sense resistor

stefan85, 2019-02-15:
Hi,

I am trying to use the TMC2660 in combination with an LPC1769 MCU. This works, but I am having issues with setting the current. With the current implementation of the TMC2130 I can specify a sense_resistor value, but this is not possible with the TMC2660. I have the feeling my current setting is a bit off. Is there a possibility for this feature to be added?

If not, what kind of value is being used by the software?

Thanks in advance.

FHeilmann, 2019-02-16:
The tmc2660 has a vsense register that can either be configured to 'low' or 'high' check the data sheet to see what these two values represent. You can then set that value in the config as is shown here : https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L987

KevinOConnor, 2019-02-18:
@FHeilmann - FYI, the tmc2130 and tmc2208 use a formula to calculate the vsense setting and current control bits from from the user provided desired amps and the ohms of the sense resistor.  I'm guessing the tmc2660 module uses a calculation specific to the duet2 wifi/eth.  I think it would be useful if we could adopt the more general mechanism to the tmc2660 module.  (It does appear the tmc2660 has a very similar mechanism of vsense/current bits that the other two drivers have.)

-Kevin

FHeilmann, 2019-02-18:
@KevinOConnor I'll look into it!

FHeilmann, 2019-03-02:
See https://github.com/KevinOConnor/klipper/pull/1338

KevinOConnor, 2019-03-11:
This should be in Klipper now (commit 4541f16c).

-Kevin

closed 2019-03-22 by KevinOConnor