Klipper3d/klipper #1260: Feature Request: Pressure Advance delay

com9, 2019-02-15:
I'm currently calibrating PA for a Bowden Extruder and having a hard time doing it. If the Extruder stops or retracts the pressure in hotend releases with a slight delay. The result is over extrusion before a corner and under extrusion after that. I suggest adding a delay which sets the pa movements slightly ahead in order to compensate for the delay. See the attached Image for reference

[image]

com9, 2019-02-15:
[image]
Here is a Photo of my Tesprints. As you can see at 0.65 I get a nice finish on the deceleration part but heavy under extrusion on the acceleration perimeter. If i set PA higher, i get underextrusion on both areas, when i set it lower i get over extrusion on the deceleration part.

cosminr86, 2019-02-15:
It looks like your square corner velocity is too low

com9, 2019-02-15:
Yes it is, I followed the Pressure Advance guide which recommends lowering corner velocity to make the PA effect more visible. But this Problem is still present at higher corner velocity and can even worse at higher speeds.

fredizzimo, 2019-02-17:
I did some similar tests a while ago. I believe the main issue is extruder slipping. Here's my theory.

A high pressure advance setting with a high printer acceleration, will cause a big velocity jumps on the extruder side. In physical terms it means that the extruder acceleration is close to infinite. The stepper motors smoothen it out a bit so it's not completely instant, and infinite.

Note that when you are tuning the pressure advance, you are essentially including the slip. With higher pressure advance values, it will still manage to push the filament further, but there will also be more slip, meaning that the pressure advance values get much higher than they should be.

So what you are seeing here, is that there is more slip when accelerating, than when deccelerating, which leads to the underextrusion on accelerating. 

Another way to see the effect of the slip is to keep the same pressure advance settings, and change the acceleration. If there's no slipping, the pressure advance should work the same way with all accelerations. But as soon as there's slipping, that's no longer the case, if you are accelerating more than the value you tuned it for, then there's even more underextrusion.

So, what can you do.
1. Lower the acceleration (and yes, it has to be much lower than you think for slip free extrusion).
2. Use s-curve acceleration, or some other jerk limited acceleration scheme that makes the printer acceleration continious, which means that the extrusion speed is continious. 

    Unfortunately the experimental smoothstep based acceleration, does not really limit the jerk, which means that in practice it can still be a very steep extrusion acceleration ramp, so it doesn't deal with all cases. So I have been working on another scheme, which is still in theory and prototyping phase, but will kill many flies in one. At least acceleration limiting on the extruder side, but probably G2 smooth. No more fake "jerk" or junction deviation, which causes discontinous velocities. No strange accel-to-deccel limits. And most importantly frequency band elimination, to greatly reduce ringing effects.

KevinOConnor, 2019-02-18:
FYI, I did try adding an "extruder lag" control about a year ago.  In my local tests (and with one other user), I didn't see an improvement.  Unfortunately, the code complexity to implement this is high (the current code generates steps for each move, but this type of control would requires generating steps not in sync with each move start/end time).  I do believe that starting extruder movements and retractions ahead of the nominal move start time would improve pressure advance, but I think it will need a more complex formula than a simple "time lag" setting and it will need some significant code refactoring.

All that said, from your pictures, it appears to me that you should continue to increase the pressure advance value (eg, to a value higher than .65).  I'd also be sure you're not tuning to the corner where the Z lifts occur.

-Kevin

com9, 2019-02-20:
I wasn´t able to go any higher than .65 bc my Extruder couldn´t keep up. But I ditched the Capricon tube and switched back to a standard PTFE tube. Now I can crank up PA up to 1.0 without Extruderproblems. The Testcubes now look a lot different, no under extrusion anywhere. But the Extruder clicking is driving me nuts already, I guess for a Bowden configuration an Extruder without transmission would work better (and more silent).

@fredizzimo I´ll try the S-Curve maybe, sounds like it could solve some of the harsh retracts.

PS: It´s not possible to set the Extruder acceleration right? Not that I'm missing anything.

jakep82, 2019-02-20:
> PS: It´s not possible to set the Extruder acceleration right? Not that I'm missing anything.

You can set max_extrude_only_accel, but that will only affect retractions.  The extruder acceleration during a print is calculated to match the movement of the printer.  

My current solution is using a very low max_accel_to_decel (500), and a faster square_corner_velocity (10).  This allows me to run higher PA values and get reasonable results with bowden.  My last few prints were good enough that I've considered staying with bowden, but at some point I'll probably switch to a direct extruder to avoid these issues.

fredizzimo, 2019-02-20:
I was referring to slowing down the general printer acceleration, which should reduce the magnitude of the instant speed changes at the extruder. But like I mentioned, you need to set it low, most likely to 500 or even less.

And yes, both the accel to decel and square corner velocity can also help for some situations.

KevinOConnor, 2019-03-11:
If there are no further comments, I'm inclined to mark this as inactive.

-Kevin

closed 2019-03-22 by KevinOConnor