Klipper3d/klipper #1261: STM32F103 ADC issue.

nowiresza, 2019-02-15:
Having a bit of an issue with the temperature values when both Extruder and Bed thermistor is enabled. When everything is at ambient temperature everything seems normal but when I heat up the extruder the temperature reading of the bed also seems to increase but the bed heater itself is not heating up. The value goes up way faster than my bed is able to heat up anyway. Also, as the extruder gets to higher temps the readings become less accurate and underleads badly. If I disable the heated bed in my config file the extruder temp reads normal and temperature values are exactly what they supposed to be. I can't seem to use more that one adc input then this issue presents itself. Possibly something off with the ADC conversion? My hardware test setup is a stm32f103 bluepill connected to a Ramps 1.4. I've confirmed the ramps is working correctly on a atmega2560 board.

Thanks in advance for any assistance.

[klippy.zip](https://github.com/KevinOConnor/klipper/files/2870910/klippy.zip)

nowiresza, 2019-02-15:
One more note. If I set the Extruder temp to 180c for example as in my log file, it struggles at the last few degrees. if I firmware_restart at that point with the bed sensor disable the actual extruder temps will show over 270c.

KevinOConnor, 2019-02-19:
I've added this to my list of things to look at.  However, I don't know when I'll be able to get to it.  Maybe @yschroeder or @grigorig will have some ideas.

-Kevin

JustAnother1, 2019-02-19:
That seem to be two issues:

- The misreading of the bed temperature:
Both sensors are probably connected to the same converter. Between measurements the converter has to switch inputs. If the next measurements is done too fast then charge from the last input may remain and lead to the described effect. There is a setting to slow down the time that the switchover takes. That value needs to be increased.

- The not reaching 180 / but actually being at 270:
The Converter  measures the signal in several steps. It needs some time to complete all the steps. If the value is read before enough time has passed a partial result may be read. This might be accurate with low values but incorrect with high values. The solution is to wait a bit longer for the result to arrive.

In both cases the issue seems to be that the ADC is configured to be to fast and therefore faster than it is able to operate. slowing things down should solve the issues.

nowiresza, 2019-02-20:
@JustAnother1 

Thanks for your help. I managed to fix the problem be change the sampling time in adc.c file from LL_ADC_SAMPLINGTIME_239CYCLES_5 to LL_ADC_SAMPLINGTIME_71CYCLES_5. Seem 239 sampling time is way too long to wait for conversion I guess. 71 could possibly still be reduces. I'll test.

closed 2019-02-21 by nowiresza