Klipper3d/klipper #1263: Printer is not ready

disco22522, 2019-02-16:
I am running my Ramps 1.4 with ATMEGA 2560 on it, running the ramps.cfg provided only editing the printer dimensions. The thing I have been messing around with, is 

"serial = /dev/ttyACM0"
I have been attempting to change it to USB04, according to Octoprint onto which port the printer controller is plugged into. Klipper has flashed successfully, but I am unable to get "status" in the terminal. It says Klipper is not ready, and no other USB definition seems to be working.

[klippy.log](https://github.com/KevinOConnor/klipper/files/2871583/klippy.log)

disco22522, 2019-02-16:
As previous posts have suggested, I found my device ID "usb-1a86_USB2.0-Serial-if00-port0" and as 

"serial: /dev/serial/by-id/usb-la86_)USB2.0-Serial-if00-port0" is not not working either.

alfsoft, 2019-02-16:
> serial: /dev/serial/by-id/usb-la86_)USB2.0-Serial-if00-port0
Did you type this line by hand? I see at least two typos here: "I" instead of "1" and extra ")" before "USB".

disco22522, 2019-02-16:
I copied it from Putty, it should actually be "usb-1a86_USB2.0-Serial-if00-port0" I will attempt to use this ID in case the characters changed from coping and pasting it.

disco22522, 2019-02-16:
So, what I ended up doing was unhooking my mega2560 from the ramps board, hook it directly up to the Pi3, and I was able to flash it, as it was finally showing under lsusb. 

Now the issue I'm battling is more hardware related, my bed thinks its at 200C - and under marlin it was showing issues. Anyways, I believe this can be closed now.

disco22522, 2019-02-16:
So I was able to get it working, by just plugged the ATMega 2560 in without the Ramps 1.4 hat, and it was actually reading ambient temperatures from the thermistors, but now it is only reading T0, at 0C and I'm still having connection issues with klipper. Here is a new log, 
[klippy.log](https://github.com/KevinOConnor/klipper/files/2872310/klippy.log)

disco22522, 2019-02-16:
Wow ! Got it working by swapping printer.cfg to run on USB0. My hypothesis is that my last mega2560 just wasn't flashing properly, finally got this spare to flash properly, and now I am getting green across the board.

Thanks for this project, excited to finally start using it. Ticket may now be closed.

closed 2019-02-25 by KevinOConnor

varunmehta, 2020-08-12:
Adding a comment here for anyone else who comes along and faces this same issue. Followed the chance @disco22522 did and got my ender 3 working again. 

Ender3Pro: 1.1.4 board 
Flashed from Marlin 1.1.9 to Klipper, everything was successful, but the printer did was not getting connected. 
`ls /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0`was found and configured in printer.cfg.

When server was started, it kept giving me errors around 
```
Starting serial connect
Unable to open port: [Errno 2] could not open port /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0: [Errno 2] No such file or directory: '/dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0'
MCU error during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 145, in _connect
    self.send_event("klippy:mcu_identify")
  File "/home/pi/klipper/klippy/klippy.py", line 222, in send_event
    return [cb(*params) for cb in self.event_handlers.get(event, [])]
  File "/home/pi/klipper/klippy/mcu.py", line 594, in _mcu_identify
    raise error(str(e))
error: Unable to connect
Build file /home/pi/klipper/klippy/../.config(1369): Wed Aug 12 02:21:50 2020

Unable to connect
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer
```

Switched `mcu` under printer.cfg to `ttyUSB0` , was able to connect and printer was back in action. 
```
[mcu]
serial = /dev/ttyUSB0
```
Interestingly, `/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0`is a symlink to `/dev/ttyUSB0` Should we use the actual `ttyUSB0` config instead ?

BlackStump, 2020-08-12:
[It is worth reading the frequently asked questions](https://www.klipper3d.org/FAQ.html#wheres-my-serial-port)