Klipper3d/klipper #1266: MCU 'mcu' shutdown: i2c timeout

ShohninDmitriy, 2019-02-16:
constantly at the beginning of printing error "MCU 'mcu' shutdown: i2c timeout
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown"
[klippy.log](https://github.com/KevinOConnor/klipper/files/2872058/klippy.log)

[printer.txt](https://github.com/KevinOConnor/klipper/files/2872062/printer.txt)
 My board BIGTREETECH SKR V1.1

KevinOConnor, 2019-02-18:
This looks like line noise on the connection to the SSD1306 display.  This isn't a common display in use - best advice I have would be either not use the display, or make sure the cables are short and of high quality.

-Kevin

ShohninDmitriy, 2019-02-18:
Prior to this was a Board MKS GEN 1.4, no problems replaced BIGTREETECH SKR V1.1 began to pour error

DanielJoyce, 2019-02-19:
Then probably a board problem...

cosminr86, 2019-02-19:
Off-topic: I had a lot of problems with skr board. It had died after 2 days

ShohninDmitriy, 2019-02-20:
[image]
At the beginning of printing with I2c screen, wild chip loading under 90%. When printing without a screen 5-7%. Why such a wild chip load. The same G code file was printed.

closed 2019-02-22 by ShohninDmitriy

Blueblue74, 2021-11-03:
I had this issue with a ssd1306 and an SKR V1.4. I found out the solution for me was to remove another I2C sensor I had on the rpi mcu, and never encountered an I2C timeout error again. Maybe klipper was messing up the two I2Cs on two different mcus or something.