Klipper3d/klipper #1267: Multi  stepper extruder

KostteT, 2019-02-16:
Hi all, and a separate hello to Kevin! I try to test "bowdirect" system for extruder on my delta printer, i use tmc2130 + nema 0.9 degree motors for standart bowden + nema 8 20*20 1.8 degree 55 grams on the direct part extruder.
How can I configure two motors for single extruder without changing/intervention python scripts? Maybe there is a way to fix it with little blood? eg [stepper_z1] -> [extruder1]... 
The method parallel connect without separate configuration does not fit and this "shameful decision", i use different diameters drive gears besides different step motor, and may be different drivers or disable stealchop on slowly speed.

Ps - Pressure_advance, thanks to him, but on the bowden 60mm over 250 speed value PA ~ 1.5 and higher - filament frays. And on every speed PA value different. When slicer will learn to count PA for each speed or maybe in klipper add table PA-speed this problem solved. But other troubles will remain.

KevinOConnor, 2019-02-18:
In order to use two stepper motors for an extruder it would require software development in the python host code.

-Kevin

KostteT, 2019-02-18:
Ok, then I'll start doing this.

sapell, 2019-09-15:
Hi, I have a dual stepper extruder (single filament) design that I'd wish to test out. It would be beneficial to have the step distances separately adjustable for the steppers as the drive gears do not have exactly same diameter. Has @KostteT come up and contributed a solution to this issue? Othervise, could @KevinOConnor or someone else give some quick pointers for needed changes to the host code to allow driving two stepoers in concert for an extruder? It wouldn't necessarily be too involved change as the same functionality is already supported for axis steppers and multiple extruders are also supported. To me it seems that a extruder object (and config section) would have an attribute to select whether e.g. the [extruder1] is only a motor or a secondary extruder altogether. Or have a motor class (and multi motor?) to be assignable to an extruder (or axis). Or is the much more to it than that?

KevinOConnor, 2019-09-15:
FWIW, I suspect the code would have a general overall structure similar to PR https://github.com/KevinOConnor/klipper/pull/1941 .

-Kevin

sapell, 2019-09-17:
Thanks for the tip Kevin! I'll dig into that and let's see if I can get my head around it.

sapell, 2019-10-01:
@KevinOConnor or anyone willing to help. I'd really appreciate a initial push to get started with implementing this feature. I'm very new to Python, but do have prior experience in OOP in general, although I'm not a programmer by profession.

What would you see to be the minimum effort way of implementing this feature? Quickly thinking it would not seem so complex task to define a 'multi_stepper' much in the way of multi_pin, but the extruder block in printer.cfg refers directly to pins and not a 'stepper' block which is a bit confusing as there is a stepper module also, but not in the context of the config file. In fact, I would be a great start to have two extruder steppers just stepping simultaneously with the same step distance, but having their own step, enable and dir pins defined. Actually I was hoping to utilize the multi_pin feature for this at first, but it's not supported in the case of stepper pins.

Should I be looking into writing new extruder kinematics, a multi stepper module, both or something else? I'd like to make as little if any modification to other modules, but I belive some changes are inevitable, right?

My idea behind this dual motor extruder is that I could use 0.9 deg steppers for extrusion, the second motor compensating the reduced torque vs. 1.8 deg stepper. Also having two filament drive gears opposite each other with a couple of guiding bearings seems to be a robust drive mechanism for the filament.

Thank you for your time and patience!

sapell, 2019-10-09:
I implemented this feature, but I'm not completely satisfied how I had to do it.
Due to the extruder kinematics module being statically referenced by the toolhead module, I resorted to modifying the original extruder module and adding an extras module 'extruder_stepper' for fetching the config section describing the added stepper.

The extruder.py was modified by changing the instance attribute `stepper` to a list containing steppers and naturally changes were needed also to `motor_off` and `move` methods. I also added a 'private' method called at `__init__` by the name of `_add_stepper`, that tries to load `extruder_stepper` module and add its stepper object to the stepper list, but only if it is defined in config to bind to that particular extruder.

My question to @KevinOConnor is that should I even bother with a pull request with the described implementation? Furhter, do you see if the extruder module coupling could be reduced by refactoring the code? I had in mind a similar approach as with axes/rails which have their steppers defined in separate config sections whereas the stepper(s) for extruders are defined in the scope of the extruder section. I understand that there propably are valid reasons for the current implementation that I just can't think of now and that to change that in a controlled way may be challenging. Currently it does not seem to be that straight forward to implement a new kinematics module (referring to the delta_mfe kinematics example) for an extruder as it is for actual kinematics. I'm open to all tips and pointers if I'm completely off track with this thinking.

Best regards, Simo Apell

KevinOConnor, 2019-10-09:
> The extruder.py was modified by changing the instance attribute stepper to a list containing steppers and naturally changes were needed also to motor_off and move methods.

FWIW, that's how I'd do it.

> I also added a 'private' method called at __init__ by the name of _add_stepper, that tries to load extruder_stepper module and add its stepper object to the stepper list, but only if it is defined in config to bind to that particular extruder.

FWIW, I'd have made a public `add_stepper()` method that just adds a stepper motor to the extruder's internal list.  Then an extruder_stepper "extras module" could read its config and call that new exposed add_stepper() method to get its stepper registered.

> My question to @KevinOConnor is that should I even bother with a pull request with the described implementation? 

Having a pull request would likely make it easier to discuss.  Be aware though, that Klipper is going into a feature freeze for the upcoming v0.8.0 release.  Also, the extruder code will be changing as part of PR #1997.  So, no harm in opening a pull request, but be aware we're several weeks out before merging something like this.

-Kevin

sapell, 2019-10-09:
Thanks for the quick response Kevin!

> FWIW, I'd have made a public add_stepper() method that just adds a stepper motor to the extruder's internal list. Then an extruder_stepper "extras module" could read its config and call that new exposed add_stepper() method to get its stepper registered.

Actually that was my original approach, but I considered this easier (lazier) way to make certain that the extruder object is already instatiated. Nonetheless, I very much like and share your view and will look into that approach again.

It'll propably take a couple of days to make the pull request as I only have so little time in the evenings and I am not most self-confident with GitHub. Personally it is not at all urgent to get this feature merged, but eventually I'd like to see that happen, maybe some others find it useful too.

-Simo

KevinOConnor, 2019-10-11:
> make certain that the extruder object is already instatiated

FWIW, you could call the add_stepper() method from a "connect" event handler.

-Kevin

sapell, 2019-10-13:
> call the add_stepper() method from a "connect" event handler.

The same occurred to me and I changed the implementation accordingly.

KevinOConnor, 2020-01-11:
FYI, basic support was merged for this (commit 90bc1679).

Thanks everyone.
-Kevin

closed 2020-01-23 by KevinOConnor