Klipper3d/klipper #1269: [FR] SAM4S (Duet Maestro) Compatability

shoreliner, 2019-02-16:
I've read through some of the prior requests for SAM4S Compatibility.  If anyone would like to tackle this functionality, I can act as a direct resource. (I am an M3D Engineer, co-developed the Duet Maestro).  The MUX Is a problem but I believe I can assist in getting around it.

If there is any interest in getting this board running, I will gladly assist.

shoreliner, 2019-02-16:
Hey @KevinOConnor  -- hope you're well my friend. =) It's been a while!  As stated above, if there is interest here I can facilitate.

sychan, 2019-02-16:
Awesome! I ordered a Duet Maestro from M3D on Valentine's Day and was wondering if any progress was being made on Maestro support.

sychan, 2019-02-16:
@grimpster Looks like there's hope for Klipper on the Duet Maestro.

Grimpster, 2019-02-17:
@davidh it was my impression last year that you had it all set for klipper, atleast that's what you made me belive on discord.

@syschan nah I gave up tbf and moved on.
I enjoyed being a part of Klipper from the beginning, but I settled with RRf for the time being.

shoreliner, 2019-02-17:
@Grimpster On some of my dev boards in that time frame there were no issues that would have prevented Klipper from running, and that is what that information was based on.  A design change required us to go to a UART MUX for motor drivers prior to the full release.  Hence why i've returned here to attempt to bring the functionality back.

sychan, 2019-02-17:
Once my Duet Maestro shows up, I'd be happy to help out. I can probably still semi-productive on a C codebase, but most of my experience is with network services and embedded interpreters, not device level stuff.

FHeilmann, 2019-02-17:
At first glance this looks like it can be implemented on the python side without having to dive into the c code.

KevinOConnor, 2019-02-18:
FYI, I did purchase a Maestro a couple of months back, and I do have a general plan to add support for the UART mux.  I don't have a timeframe to complete that though.

-Kevin

shoreliner, 2019-02-18:
@FHeilmann I would assume that's correct, outside of the UART mux issue.  I haven't yet had any time to delve into it.  Communication to the board should be the same, I plan to try establishing over USB and begin testing when I have some free time.

@KevinOConnor Good to hear; it's a nice board.  If you need any resources, please reach out. =)

--DH

shoreliner, 2019-02-18:
@FHeilmann Also, if taking this on is of interest to you and you need hardware resources, I should be able to help you.  Just let me know.

FHeilmann, 2019-02-18:
@shoreliner thank you for offering this but at present I am kneedeep in the atsamd port so sending me hardware would likely not yield any quick results I'm afraid.

- Florian

bwright85428, 2019-02-18:
I have started looking into working out support for the UART Mux on the maestro. I haven't been too much involved with Klipper until recently. @FHeilmann you said that you think the implementation might be possible with Python only, is there anymore information that you might be able to provide without spending too much time digging into it? @KevinOConnor if you could provide your preemptive plans as well that might help guide me to the right place.

FHeilmann, 2019-02-18:
@bwright85428 basically my 'idea' was an extra which takes as configuration 3 pins and then exposesd an interface which translates a single integer (e.g. 6) to the individual pin states (e.g. 0 1 1). This would then have to be called before the UART transaction is made by the extra that controls the TMC222x. I haven't thought about it further.

KevinOConnor, 2019-02-19:
FWIW, I planned to modify klippy/extras/tmc2208.py to set the desired mux pins prior to transmitting on the uart.  That's actually pretty straight forward to do.  Unfortunately, the Maestro does not have separate enable pins for each stepper driver, so we really want to perform the stepper enable via a uart command.  Getting that right requires a bit more work, and as a result I haven't started any of the work yet.

-Kevin

bwright85428, 2019-02-20:
Do the steppers have to be enabled before interfacing with the drivers? I am not able to read the registers on the drivers and guessing that the steppers need to be enabled power on the drives.

KevinOConnor, 2019-03-11:
I'm inclined to mark this one as inactive.  I'd like to get to it myself, but don't know when that will happen.  If someone else is actively working on it, then let me know.

-Kevin

maxcom99, 2019-03-11:
> I'm inclined to mark this one as inactive. I'd like to get to it myself, but don't know when that will happen. If someone else is actively working on it, then let me know.
> 
> -Kevin

Oh NO! @shoreliner any chance you can jump on this so i can actually use my Maestro that is sitting here simply because it doesn't support Klipper? Thank you in advance

bwright85428, 2019-03-13:
@KevinOConnor I have been working on this with no luck. I get IFCNT register read errors during start up Manually setting the mux output pins in the printer.cfg. Not sure why, I can post the Klipper log if you want to take a look. If there is somewhere I can get more technical help while asking specific questions, I’d love to keep pushing this. 

It looks to me like tmcuart.c isn’t picking up anything. It outputs read=“” after tm2208 sends a tmcuart send with response request. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

KevinOConnor, 2019-03-14:
If you have technical questions on the code, this is probably the best place to ask them.  However, note that I haven't had time to look at the analog mux yet, so I don't know what is needed or why it would not work.  If you have specific questions on the code, I can try to help with that.

Yes, in general, all tmc drivers need to have motor power enabled in order to respond to queries.

-Kevin

bwright85428, 2019-03-19:
@KevinOConnor I have attached my klippy.log file just in case you would like to look at it. I tried pulling the new master down today and use the gpio output pins at MCU startup but I am still having the same issues. I can't think of anything else to try as I am not even trying to switch the mux at this point, I am just trying to set the first port (all 3 inputs on the mux set to low) and configure 1 tmc2208 driver for testing. Let me know if there is anything that you can think of for me to look into and I'll give it a go. 

As for the maestro's enable, from what I can see all of the drivers on the main board share an enable. I have that set and I have 24VIN on the board. 

[klippy.log](https://github.com/KevinOConnor/klipper/files/2981287/klippy.log)

hannesweisbach, 2019-04-05:
I wanted to get a Maestro for my printer to use with Klipper. Unfortunately it doesn't seem to be supported at the moment. Is the SN74CB3Q3251 mux the only thing in the way of supporting the Maestro or is there more work to be done? Implementing the mux doesn't seem too complicated, but I'd like to do it in a way you feel comfortable merging the code.

> FWIW, I planned to modify klippy/extras/tmc2208.py to set the desired mux pins prior to transmitting on the uart. That's actually pretty straight forward to do. Unfortunately, the Maestro does not [...]

I'm assuming by "uart command" you mean a command from the host (RPi) to the Duet (Sorry, new to Klipper). I guess you want to have a python object which you give a number and that sets the (in this case 3) output lines. You'd add the mux to the config and have the stepper class(es) see if there is a mux, and if so, use it?

How would like the config to happen? Abuse the cs_pin config option, or add a new mux-something option?

hannesweisbach, 2019-04-05:
> [klippy.log](https://github.com/KevinOConnor/klipper/files/2981287/klippy.log)

@bwright85428 I guess you're using the wrong pins. If I read that log correctly you switch PA14, PA16 and PA17:

```
[output_pin mux0]
pin = !PA14

[output_pin mux1]
pin = !PA16

[output_pin mux2]
pin = PA17
```

But the mux is connected to P**C**14, 16, and 17

shoreliner, 2019-04-05:
Hey folks -- I wish I had the time myself to assist directly right now, and may this weekend.  Just been super busy with some other things lately.

Anyways, I dug this out of some of our comments from the original implementation of the UART mux -- it should help you. Keep in mind this is wrtten in c++

`// UART interface to stepper drivers
Uart * const UART_TMC22xx = UART0;
const IRQn TMC22xx_UART_IRQn = UART0_IRQn;
const uint32_t ID_TMC22xx_UART = ID_UART0;
const uint8_t TMC22xx_UART_PINS = APINS_UART0;
#define TMC22xx_UART_Handler	UART0_Handler

// Define the baud rate used to send/receive data to/from the drivers.
// If we assume a worst case clock frequency of 8MHz then the maximum baud rate is 8MHz/16 = 500kbaud.
// We send data via a 1K series resistor. Even if we assume a 200pF load on the shared UART line, this gives a 200ns time constant, which is much less than the 2us bit time @ 500kbaud.
// To write a register we need to send 8 bytes. To read a register we send 4 bytes and receive 8 bytes after a programmable delay.
// So at 500kbaud it takes about 128us to write a register, and 192us+ to read a register.
// In testing I found that 500kbaud was not reliable, so now using 200kbaud.
const uint32_t DriversBaudRate = 200000;
const uint32_t TransferTimeout = 10;				// any transfer should complete within 10 ticks @ 1ms/tick

constexpr Pin TMC22xxMuxPins[3] = { 50, 52, 53 };	// Pins that control the UART multiplexer, LSB first
`

I'll try and focus this weekend and give you something more solid.

micahn, 2019-04-11:
i just purchased a maestro before reading this thinking all duet 2's were supported lol. once it gets here(probably a week or so) i am fully open to helping with this if needed by then. not having klipper may induce tears.

KevinOConnor, 2019-06-13:
FYI, I finally got around to implementing the TMC2208 analog mux support on the Maestro (commit f6864da5).  I haven't yet printed anything on the Maestro, but basic functionality seems to work.  I also went ahead and committed a sample config file for the Maestro (commit e697fc11).

-Kevin

Grimpster, 2019-06-13:
=O
As soon as I'm done printing all my abs upgrades I'll give it a run!
Good news!

KevinOConnor, 2019-06-28:
Lets close this issue.  If anyone has issues with the Maestro, please open a new issue with a Klipper log of the event.

-Kevin

closed 2019-06-28 by KevinOConnor