Klipper3d/klipper #1270: Error on X-homing: Endstop x still triggered after retract

Camaro1, 2019-02-16:
Hi everyone,
I'm trying to set up Klipper on my CoreXY Printer with TMC2130 on X, Y and Z. I used the Sensorless Homing feature for X and Y before with Marlin and it worked alright. So I think it's some kind of configuration issue.
When homing X I get the message "Recv: !! Endstop x still triggered after retract" which then obviously terminates all further homing and X is not "homed".(no movement in that direction) I don't have this issue when homing Y or Z. (However Z uses a probe)
As a test I tried the "default" homing without [homing_override] but this showed the same result.
As I'm on a CoreXY i connected the diag1 pin of the X and the Y dirver together to all four endstop pins (X+, X-, Y+, Y-), which worked fine with Marlin. However I'm not sure if this is the way to do it, but homing Y works with this setup.
Using a higher sensitivity (driver_SGT: 3) causes the print head to not fully home.

If you need any more information, just let me know.
[klippy.log](https://github.com/KevinOConnor/klipper/files/2872361/klippy.log)

Thanks in advance for your help.
Camaro

jakep82, 2019-02-16:
Set homing_retract_dist to 0 for X and Y.

Camaro1, 2019-02-16:
This was fast! Thanks for your help. Homing the X axis works now.
Could you explain, why this worked? Or the other way round: Why doesn't homing with retract work?

cmock, 2019-02-16:
I don't think you're supposed to tie the diag/endstop pins together. 

On my CoreXY, I have wired them separately, and then sensorless homing was just an issue of finding the right value for driver_SGT according to the docs. I am homing with the default 5mm retract.

jakep82, 2019-02-17:
Yes, I don't think it's wise to tie them together.  I've never had success on my printer using anything other than a retract value of 0.

Camaro1, 2019-02-17:
Ok, thank you!
My intention behind wireing them together was: On CoreXY both motors move for a X or Y move. So both motors should stall at the same time. But if i try to home X and the Y motor stalls a little earlier I wanted the movement to stop immediately and not only when the X motor stalls.
My wireing may nevertheless be wrong though.
Maybe I'll rewire it as soon as I have time.
But the initial issue is solved. Thanks for your help.

closed 2019-02-17 by Camaro1