Klipper3d/klipper #1271: Ramps 1.4 E1 for Z1

disco22522, 2019-02-16:
I am currently setting up the firmware to run properly with my Ramps 1.4 setup, and I am running into an issue where my z-axis is binding, and while my stepper driver is set to v1 with the potentiometer, It is making a loud noise and binding. So, I am wondering, if I can use Ramps 1.4 E1 to setup for my Z1 - I kind of know where to start, 

[stepper_z1]
# Z1 Stepper - Rear Left
step_pin: z:ar60
dir_pin: z:ar61
enable_pin: !z:ar56
#   Y on mcu_z
step_distance: 0.00250
#   400 steps per mm - 1.8 deg - 1/16 microstepping

Although, I am not sure these pins reflect what pins I have on Ramps 1.4, and I cannot seem to find any reference out there, that shows me what pins E1 is. Any assistance is appreciated.

wizhippo, 2019-02-16:
The example ramps confit show e1 to be the below.  Maybe your pins follow the same. 

#[extruder1]
#step_pin: ar36
#dir_pin: ar34
#enable_pin: !ar30
#heater_pin: ar9
#sensor_pin: analog15

disco22522, 2019-02-17:
Thank you Wizhippo, ar36 seems to be the correct pin.

The issue I am running into now, are loud stepper motor noises coming from the Z-Axis. I have tried swapping out stepper drivers, and it doesn't seem to be a hardware issue. The max homing speed I have set is 2mm/s, and acceleration of 50 with a max travel speed of 4 mm/s. The steps at .0025, which is correct for my M8 lead screws. So either .0025 is too much, which it isn't because 400 is what I had set in Marlin or the homing speeds are still too fast.

wizhippo, 2019-02-20:
Are you z1 and z moving in the same direction?  Maybe post a video so we can see what the binding looks like

AxMod3DPrint, 2019-02-24:
@disco22522 - set your homing speed on Z slow if using M8 Threaded Bar. You want to be around 3-5mm/s at most. Anything over and homing will fail.

KevinOConnor, 2019-02-25:
I'm not sure what the state of this issue is.  If there is a request for assistance, the Klipper log file must be attached - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

-Kevin

closed 2019-03-04 by KevinOConnor