Klipper3d/klipper #1275: Heater extruder not heating at expected rate - consistent failure in one specific print

galorin, 2019-02-17:
[klippy.zip](https://github.com/KevinOConnor/klipper/files/2872885/klippy.zip)

I have attached my Klippy log file.

The extruder heats, and prints just fine on a couple half-hour prints, but in this one particular the print fails after about seven minutes of printing, and always at the same exact point in the print.  I am not sure what needs to be changed to get the print to stop failing when the extruder is pretty much fine.

Trontastic, 2019-02-17:
I'm also had this problem, not sure if it's help you but maybe you can try if you also have a 3D touch:
I noticed my 3d Touch started to fail (softly blinking), I replaced that and there was no heating problem after that.

galorin, 2019-02-17:
I'm using an inductive probe, powered directly from the PSU and the signal
pun is being run through a reverse biased Zener diode, knocking it down to
5v.

KevinOConnor, 2019-02-18:
It appears your power supply is not capable of providing sufficient power to the heater to maintain the given temperature and/or the heater has malfunctioned.  (The log indicates the extruder power was fully turned on for an extended period but the heater continued to drop in temperature.)  This issue can only be fixed in hardware.

-Kevin

galorin, 2019-02-20:
I've ordered a new heater cartridge.  Will try again.  Closing this issue as it's hardware.

closed 2019-02-20 by galorin