Klipper3d/klipper #1276: output_pin -value & shutdown_value - option to "leave it" as it is

Trontastic, 2019-02-17:
Hi!

I would suggest a little feature request for the output_pin section.

I have an ATX PSU and when I restart the Klipper software it's also restart the ATX unit.
I don't want to set the shutdown_value to 1 because when the PSU is not running it would turn it on then off.

My question is that can you please add a -1 option to "value" and "shutdown_value" settings to don't change the pin at all when it's start/shutdown ?
So when the ATX is on, leave it on, when it's off, leave it off so it would not restart needlessly.

KevinOConnor, 2019-02-18:
This would require a developer to implement.  However, I don't think it is particularly straight forward to implement as, in the general case, when the software starts up it does not know what the state of the pins are in.

-Kevin

Trontastic, 2019-02-18:
> This would require a developer to implement. However, I don't think it is particularly straight forward to implement as, in the general case, when the software starts up it does not know what the state of the pins are in.
> 
> -Kevin

That would be not a problem, the important thing is to don't set the pin after boot/shutdown when I set it to -1 (or whatever).

But in an other issue TMC made this irrelevant, because I can't use it with TMC drivers so... :)

closed 2019-02-18 by Trontastic