Klipper3d/klipper #1277: ATX and TMC2130 don't like each other

Trontastic, 2019-02-17:
Hi!

I upgraded my setup a little with the PS_ON pin.
I added the ability to control my PSU from klipper/octoprint.

However when the system boots, the TMC2130 drivers won't working anymore.
G28 commands timeout and octoprint disconnect from Klipper.

I read in a Marlin issue that the TMC drivers have to be initialized again after the "real boot" (after the ATX spin up).

Is there a command to initialize the drivers?
I would add them to my M80 macro (with a delay maybe?).

My config:
[output_pin atx]
pin: !ar6
value: 1
#   The value to initially set the pin to during MCU configuration.
#   The default is 0 (for low voltage).
#shutdown_value:
#   The value to set the pin to on an MCU shutdown event. The default
#   is 0 (for low voltage).

[gcode_macro M80]
gcode:
	SET_PIN PIN=atx VALUE=1
	
[gcode_macro M81]
gcode:
	SET_PIN PIN=atx VALUE=0

CHILLYSMOKES, 2019-02-17:
Sir, the TMC2130s require power before MCU. Googalize more. You will not solve this issue with code, alone.

Trontastic, 2019-02-17:
> Sir, the TMC2130s require power before MCU. Googalize more. You will not solve this issue with code, alone.

I'm not sure what are you talking about, but I also said that the TMC need power to initialize.
When the system boots, the GenL does not have enough power to do it, that's why I asked that is there any TMC_INIT gcode I can use after I turn on the ATX or any other method.

cmock, 2019-02-17:
The datasheet for the TMC2130 says Vm (the motor voltage, 12V from the ATX in your case) needs to be on before the 5V supply. The reason is that Vm feeds the internal voltage regulator that provides power to the internal logic of the TMC2130.

You could get Stepstick Protectors from Watterott, those have a diode from 5V to Vm, or you could put in a diode yourself.

The remaining problem then is that when you somehow try to run the motors (or anything on 12V) when your ATX power supply is off, it will draw power from the 5V rail and probably shut down the microcontroller.

CHILLYSMOKES, 2019-02-17:
It's not just a coding issue. The TMC2130s require power first. So if you are using a ramps/mega sandwich the ramps needs power before the mega; long enough for init. This is to protect the chips... unless you have the standoffs, chip protectors from watterott. If you continue to power the sandwich using the PS_ON pin with some kind of code without staging or the chip protectors you will eventually have a bad day. You will not solve this issue with code alone, sir.
I and others have broken the powering, between the sandwich layers, into stages using relays switched from the Pi's gpios and controlled directly through Octprint. At a minimum, IF you really want to use the PS_ON pin, you need the standoffs... and some nice person that knows TMC/klipper coding. 
Though, it seems dangerous to code a re-init when other things need to be in place.

KevinOConnor, 2019-02-18:
FYI, the only way to do currently resend the tmc init is to issue a RESTART command.

-Kevin

closed 2019-02-18 by Trontastic

CHILLYSMOKES, 2019-02-21:
Oops somebody just added some code for this.

thucar, 2019-02-24:
Well great, guess I need to grab me some standoffs. For the time being, at least I know to issue a RESTART after powering on to keep things running. Thanks for pointing me in the right direction, @CHILLYSMOKES