Klipper3d/klipper #1280: Attempting to enable M204 as a gcode_macro causes connection failure without explanation

fxbisto, 2019-02-18:
I typed up a fresh printer.cfg and attempted to include the following:

> [gcode_macro M204]
> #	Allows M204 S[value] to manually specify acceleration during gcode.
> #	Usage: M204 S[acceleration (mm/s2)]
> #
> ;default_parameter_S=1200
> gcode:	SET_VELOCITY_LIMIT ACCEL={S} ACCEL_TO_DECEL={S}
> 

After doing so I spent the next 45 minutes trying to work out why Klipper would simply reply:

> Internal error during connect.
> Once the underlying issue is corrected, use the "RESTART"
> command to reload the config and restart the host software.
> Printer is halted

Is there some way to be verbose about this kind of error to save head scratching in future?

cosminr86, 2019-02-18:
M204 gcode is already built in Klipper

fxbisto, 2019-02-18:
Yes I was hoping to override the default behaviour as it doesn't appear to change accel_to_decel at the same time?
For example: setting  accel=2400 accel_to_decel=1200 then using M204 S1000 gives a console output of accel=1000 accel_to_decel=1200

Hywelmartin, 2019-02-18:
I can understand that kidnapping an existing gcode would generate Error.
I have read thru the kinematics docs and can't say that I'm certain about how Klipper handles retardation per move.. For acceleration it's more clear.
Lets say that I have set up max acceleration to 5000mm/s2 in printer.cfg. then retardation will be 2500 mm/s2 per default.

Then in the gcode I let the slicer "handle" the acceleration per typ of feature. Will the retardation be half of the M204 set value or will it be 2500 mm/s2 or the M204 set value

The Klipper gcode is (in my thinking) a limiter, not a set point

> SET_VELOCITY_LIMIT [VELOCITY=<value>] [ACCEL=<value>] [ACCEL_TO_DECEL=<value>] [SQUARE_CORNER_VELOCITY=<value>]

Any one with more insight ...

fxbisto, 2019-02-18:
As you said:

> kidnapping an existing gcode would generate Error.

I'm comfortable with that, although it would be desirable to be able to override the functionality.

The main issue is that Klipper couldn't identify this failure as a parsing error or gcode definition conflict, just gives a generic failure without identification to the point of error.

cosminr86, 2019-02-18:
Afaik accel_to_decel always will be half of acceleration if not set otherwise.

Anyway I don't think cura can have other gcode output than m204 s(value)

Maybe you can try to play with a custom gcode using jerk settings in cura.

fxbisto, 2019-02-18:
Perhaps that's the case (it would seem logical) however when a value for accel_to_decel is not explicitly specified the console reports that it is set to half of max_accel still even when M204 changes accel to another value.
I am currently using M205 Jerk control in Cura to adjust square_corner_velocity per model

KevinOConnor, 2019-02-18:
Thanks.  The error reporting should now be improved (commit b260eb91).

FYI, the max_accel_to_decel that is reported by SET_VELOCITY_LIMIT is the value in use (the only caveat is that if max_accel_to_decel is greater than max_acceleration then max_accel_to_decel is effectively disabled).  The max_accel_to_decel does not change on a SET_VELOCITY_LIMIT command unless one explicitly sets a new value for it.

-Kevin

fxbisto, 2019-02-19:
Hi Kevin,
Thanks for the clarifications regarding accel_to_decel, much appreciated.
You're a gent

closed 2019-02-19 by fxbisto