Klipper3d/klipper #1281: 2 printers in kllipper

shadrincev, 2019-02-18:
Good afternoon. I have 2 printers, each with the same bigtreetech SKR 1.1 Board. On the first: is bltouch, TMC2208 uart and bed_mesh for the table. And on the second: inductive sensor, simple drivers A4988 and bed_tilt for the table. How can I connect two printers at the same time and what to register in the printer.cfg? Or someone has already implemented the connection of two printers, can share an example printer.cfg? Sorry for the bad English, used the translator.

KevinOConnor, 2019-02-18:
If you are looking to control one physical printer with two control boards then take a look at config/example-multi-mcu.cfg .  If you're looking to control two separate printers with one Raspberry Pi, then take a look at the FAQ - https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#can-i-run-multiple-instances-of-klipper-on-the-same-host-machine

-Kevin

shadrincev, 2019-02-18:
Thank you for the link to the FAQ, but I thought there is a ready script to run. I do not have the necessary knowledge to change the script to install.

shadrincev, 2019-02-19:
Good day everyone. Like I said, I'm not good at writing scripts. But, read the FAQ and tried to change install-octopi.sh. 
```
Instead of the KLIPPY_EXEC=${PYTHONDIR}/bin/python
KLIPPY_ARGS="${SRCDIR}/klippy/klippy.py ${HOME}/printer.cfg -l /tmp/klippy.log
```
I ordered 
```
KLIPPY_EXEC=${PYTHONDIR}/bin/python
KLIPPY_ARGS="${SRCDIR}/klippy/klippy.py ${HOME}/printer2.cfg -l /tmp/klippy2.log -I /tmp/printer2
```
and saved under the name 123.sh. 
After installation install-octopi.sh and installation klipper I start on performance 123.sh it is executed and in the directory /home/pi/ there is a file printer2. In folder tmp file klippy2.log is missing there is only file klippy.log.
Then prescribe octoprint /tmp/printer2 and trying to connect, but this does not happen until the prescribed /tmp/printer. That is, /tmp/printer 2 is not created. What am I doing wrong?

shadrincev, 2019-02-20:
Once again good time of day, then prompted that it is necessary to enter line 
~/klippy-env/bin/python ~/klipper/klippy/klippy.py ~/printer2.cfg -l /tmp/klippy2.log -I /tmp/printer2
 as a command, but it is hanging on the green cursor and then nothing happens...
[image]

shadrincev, 2019-02-20:
Thank you, for starting understand the picture above. The script in the autorun almost scored. Now the question remains on the two circuit boards SKR 1.1. They have the same vendor and id. How to separate them, so that when you turn on two printers are not assigned a different ttyACM? In make menuconfig I see USB idle, what is he responsible for? Can I use it to split the id? Or are there other ways?

KevinOConnor, 2019-02-22:
See: https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#wheres-my-serial-port

shadrincev, 2019-02-23:
Good afternoon. You misunderstood me. I have 2 printers and each is control Board SCR 1.1. When the command 
ls-l /dev/serial/by-path/ 
output two boards with the names of the ACM*. The boards have the same **idVendor** and **idProduct**. But if you turn off or turn on any of the printers, the name ACM * changes randomly. **Here is me and interested in, as fix ACM* to port if idVendor and idProduct is the same?**
[image]

AxMod3DPrint, 2019-02-23:
Each board should really have an internal unique ID on the USB chip, but Chinese stuff rarely does.  As this is at system level you can use udev to assign static ports. IE if this device is plugged into this port assign this name. 

https://msadowski.github.io/linux-static-port/

https://unix.stackexchange.com/questions/66901/how-to-bind-usb-device-under-a-static-name

shadrincev, 2019-02-23:
Yeah, I'm trying to use udev to create a rule 
/etc/udev/rules.d/99-usb-serial.rules 
SUBSYSTEM=="tty", KERNELS=="6-1:1.0", SYMLINK+="FRIBOT"
SUBSYSTEM=="tty", KERNELS=="8-1:1.0", SYMLINK+="P905" 
write there. 
I reboot the rule and the system, but everything remains the same. The rule doesn't work.

shadrincev, 2019-02-24:
Good afternoon.
The udev service is running.
[image]
But it is impossible to cling to the klipper2, it turns out the klipper2 service is not running correctly.
I will describe the order of their actions, and who can let correct where I was wrong.
Copy the autorun and klipper settings
```
sudo cp /etc/init.d/klipper /etc/init.d/klipper2
sudo cp /etc/default/klipper /etc/default/klipper2
sudo cp /var/run/klipper.pid /var/run/klipper2.pid
```
`sudo nano /etc/default/klipper2`
open the configuration and change the ARGS to change the path to the config and to the log and add at the end before the quotes -I /tmp/printer2
```
# Configuration for /etc/init.d/klipper
KLIPPY_USER=pi
KLIPPY_EXEC=/home/pi/klippy-env/bin/python
KLIPPY_ARGS="/home/pi/klipper/klippy/klippy.py /home/pi/printer2.cfg -l /tmp/klippy2.log -I /tmp/printer2"
```
`sudo nano /etc/init.d/klipper2`
open AutoPlay and all values klipper should be changed to klipper2 and add DAEMON=/usr/bin/klipper
```
#!/bin/sh
# System startup script for Klipper 3d-printer host code

### BEGIN INIT INFO
# Provides:          klipper2
# Required-Start:    $local_fs
# Required-Stop:
# Default-Start:     2 3 4 5
# Default-Stop:      0 1 6
# Short-Description: Klipper2 daemon
# Description:       Starts the Klipper2 daemon.
### END INIT INFO

PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
DESC="klipper2 daemon"
NAME="klipper2"
DEFAULTS_FILE=/etc/default/klipper2
PIDFILE=/var/run/klipper2.pid
DAEMON=/usr/bin/klipper

. /lib/lsb/init-functions

# Read defaults file
[ -r $DEFAULTS_FILE ] && . $DEFAULTS_FILE

case "$1" in
start)  log_daemon_msg "Starting klipper2" $NAME
        start-stop-daemon --start --quiet --exec $KLIPPY_EXEC \
                          --background --pidfile $PIDFILE --make-pidfile \
                          --chuid $KLIPPY_USER --user $KLIPPY_USER \
                          -- $KLIPPY_ARGS
        log_end_msg $?
        ;;
stop)   log_daemon_msg "Stopping klipper2" $NAME
        killproc -p $PIDFILE $KLIPPY_EXEC
        RETVAL=$?
        [ $RETVAL -eq 0 ] && [ -e "$PIDFILE" ] && rm -f $PIDFILE
        log_end_msg $RETVAL
        ;;
restart) log_daemon_msg "Restarting klipper2" $NAME
        $0 stop
        $0 start
        ;;
reload|force-reload)
        log_daemon_msg "Reloading configuration not supported" $NAME
        log_end_msg 1
        ;;
status)
        status_of_proc -p $PIDFILE $KLIPPY_EXEC $NAME && exit 0 || exit $?
        ;;
*)      log_action_msg "Usage: /etc/init.d/klipper2 {start|stop|status|restart|reload|force-reload}"
        exit 2
        ;;
esac
exit 0
```
reboot DAEMON
`sudo systemctl daemon-reload`
To make everything automatically run update rc.d
`sudo update-rc.d klipper2 defaults`
Start a new instance of klipper2
`sudo /etc/init.d/klipper2 start`
Check if the service was successfully added
`systemctl status klipper2.service`
the inscription is active lights up green it means the service is started.
Then I try to hook up to the printer freeboot and p905. The octoprint does not connect, even if you 
explicitly specify which port to connect to Then I try to hook up to the printer freeboot and p905. Octoprint does not connect, even if you explicitly specify which port to connect to (next to /tmp/printer2 specify /dev/fribot). Something like that. What am I doing wrong?

shadrincev, 2019-02-24:
Thank you all, got it.

closed 2019-02-24 by shadrincev

AxMod3DPrint, 2019-02-24:
Glad you got it sorted, for other's reference, what did you do?

shadrincev, 2019-02-24:
All steps described above. The only thing I forgot is sudo chmod for a copy of klipper. That's why there were problems.

shadrincev, 2019-02-28:
More details here https://3dtoday.ru/blogs/shadrincev/double-klipper/