Klipper3d/klipper #1282: Klipper doesn't wait for lower temperature in M109

guffy1234, 2019-02-18:
hi

I have end gcode that cooling printer. And it looks like klipper doesn't wait when you set lower temperature (looks like it compares only with > )

`
M107 ; Turn off fan
; Disable all extruder
G91 ; Relative positioning
T0
G1 E-1 ; Reduce filament pressure
M104 T0 S0
G90 ; Absolute positioning
G92 E0 ; Reset extruder position
M140 S0 ; Disable heated bed
M84 ; Turn steppers off
M117  cooling printer... ; show "cooling printer..."
M106 S255 ; turn on fan 100%
M109 R100.000000 ; wait hotend 100C
M106 S0 ; turn off fan
M81 ; power off
`

BlackStump, 2019-02-18:
dont think Klipper supports R parmeter 

https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md
Set extruder temperature and wait: M109 [T<index>] S<temperature>
Note: M109 always waits for temperature to settle at requested value

guffy1234, 2019-02-18:
hmm.. you are right. i didn't sight that it is R

Hywelmartin, 2019-02-18:
I'm rather sure that M109 S100 will wait  for both heating and cooling...

LazaroFilm, 2019-02-20:
I would love to get the R setting. 
I use it to probe with my nozzle hot, at at least 170c. but if I do antoher print right after it waits to cool back down to 170 before probing. R would allow to go straight to probing while it's still cooling down.

KevinOConnor, 2019-03-04:
I'm going to close this as I think the original question was answered (Klipper does not support the "R" flag in M109).

-Kevin

closed 2019-03-04 by KevinOConnor

LazaroFilm, 2019-03-07:
Any plans to implement it in the future?

stefanfoulis, 2019-09-01:
If there is interest in supporting the `R` flag in Klipper, I'd like to help. I know python, but not C. So if the logic needs to be implemented in python I could make a PR.
If so, could someone point me in the general direction of where in the source code this would be implemented?

F4HTB, 2020-05-31:
+1 for add the cooling support. :D