Klipper3d/klipper #1283: Metro m4 config

jimmyhorns, 2019-02-18:
I am struggling to work out how to use this with a metro m4 microcomtoller board. I’m aware that I am likely to need to build my own ramps type board, however I couldn’t see an example config file for this board to start working out the pin outs.

Am I missing something?...

Thanks in advance

jakep82, 2019-02-18:
As you noted, you will need to design a controller board.  The design of the board would determine the pins.  You would then need to create your own config file with pins that match the board.

FHeilmann, 2019-02-19:
the support for the samd51 is still in super early stages. Most of the things don't work properly yet, thats why there is no documentation of any sort.

jimmyhorns, 2019-02-19:
Ah right - I thought this might be the case. I shall stick with my mkr sbase for now then and try and get this working first of all :)

Thanks for your help

supermerill, 2019-02-28:
for the board, you can try with a cnc shield. You may not have enough output for everything (heaters, thermistors), so maybe a little ramps as a second mcu is needed.

KevinOConnor, 2019-03-04:
There is still ongoing work with the SAMD51 line of chips.  However, it should be possible to launch a 3d printer with it.  One can use config/example.cfg as a starting point - just be sure to use the appropriate pin names (PA0-31, PB0-31, etc.).  Which pins to place in the config file depends on which pins one used to wire the printer.

-Kevin

KevinOConnor, 2019-03-11:
I'm going to close this - as far as I know it should be possible to wire up a board with the metro-m4.

-Kevin

closed 2019-03-11 by KevinOConnor