Klipper3d/klipper #1286: Powering up MCU after RPI causes communication error

lmcbmai, 2019-02-19:
RPI, connected to my printer, is powered on most of the time, while the printer's mainboard, which has it's own 15V power supply and on/off switch, is usually powered on only during printing process. The problem is - +5V DC from RPI power supply via USB cable reached mainboard and keeps MCU, LCD and recently installed IR Z-probe on. So, I am looking for a ways to cut off this power for the time when I am not printing. 
I tried 2 approaches - command line based:
```
sudo ./uhubctl --loc=1-1 --ports=2 --action=off
sudo ./uhubctl --loc=1-1 --ports=2 --action=on
```
and using USB cable with cut +5V wire.

While both worked fine on my RPI 3B+, the problem I am having is - when MCU is powered up while RPI is up and running, Klipper fails to establish communication with it. RPI reboot while MCU is powered on fixes this problem, but I wonder if there is another way that does not require reboot of the host.

Thanks

cosminr86, 2019-02-19:
Usually a firmware_restart on octoprint terminal do the job.

lmcbmai, 2019-02-20:
I cannot execute FIRMWARE _RESET because when connection is not established Terminal does not accept any commands and all buttons on Klipper tab are disabled.

cosminr86, 2019-02-21:
So you can't connect to the /tmp/printer ?
Did you check the octoprint settings to ignore the printer errors? 

Attach your log file.

FHeilmann, 2019-02-22:
there's def. something amiss with your setup.

even if the mcu is not powered the klippy process will still be running on the pi, so you should be able to send commands to it, such as FIRMWARE_RESET as cosminr86 suggested.

lmcbmai, 2019-02-22:
Thank's to you both. I have a workaround (RPI reboot fixes it) and will try to investigate this later, when I have time. Closing this issue for now.

closed 2019-02-22 by lmcbmai

Terror-Gene, 2019-03-21:
I've always had this issue using Klipper.
Here is what I've discovered tonight after some testing:

I shutdown the printer, rebooted the RPi & once it had started up again I powered on the printer.
It connected right away with:
> Recv: // Klipper state: Ready

Then i turned the printer off & received:
> Recv: // Klipper state: Shutdown

Now if i turn the printer on it does not connect on its own anymore.
If I try a FIRMWARE_RESET I get:
> Send: FIRMWARE_RESET
Recv: // Lost communication with MCU 'mcu'
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Printer is shutdown
Recv: !! Lost communication with MCU 'mcu'
Recv: ok

The only way to make it reconnect is either:
Reboot the RPi & it will connect straight away (if the printer is already on or as soon as its turned on.)
or
Run the RESTART or FIRMWARE_RESTART command then reboot or power on the printer (wont connect if the printer is already on):
> Send: Restart
Recv: ok
Recv: // Klipper state: Disconnect
[...]
Recv: // Klipper state: Ready

> Send: FIRMWARE_RESTART
Recv: ok
[...]
Recv: // Klipper state: Ready

Seems this is related to the state of the printer being "shutdown" & it doesn't seem to reconnect unless that state is switched back to "disconnected".

Now that i have realized what the issue is i can at least make a RESTART macro with OctoKlipper & click that before turning the printer on. not ideal, but it's better than restarting the RPi every time i want to print. Hopefully this can be resolved as it was frustrating me for ages & tempted me to switch back to another firmware.