Klipper3d/klipper #1288: Heater not configured

raoulfa, 2019-02-20:
Hi, I keep getting the error when trying to print "Heater not configured" When I try to manually heat up the nozzle via Octoprint there is no Problem. Printer I am using is a Geeetech A10M.
[klippy.log](https://github.com/KevinOConnor/klipper/files/2884578/klippy.log)

ShohninDmitriy, 2019-02-20:
not in the config spelled out the thermistor pin to the extruder and the sensor type and method of heating

raoulfa, 2019-02-21:
Thanks for your answer! But what I not understand why did the PID Tuning work and why can I heat it up manually? Or am I wrong and it is not related?

Hywelmartin, 2019-02-21:
https://github.com/KevinOConnor/klipper/blob/328bd89fc11cad86802c5a35055f6b6d4a8243f0/config/example-extras.cfg#L300

I'm not really sure but everything else is already configured... in [extruder]

```
[extruder1]
step_pin = ar49
dir_pin = ar47
enable_pin = !ar48
step_distance = .002304147
shared_heater = extruder
```

KevinOConnor, 2019-02-22:
It looks like some g-code attempted to set the temperature of an extruder that wasn't configured.  If you can't find the offending g-code, then issue an M112 immediately after the error report and post the resulting log - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

-Kevin

raoulfa, 2019-02-25:
Thanks Kevin, so far I was not able to figure our the offending gcode. Here my log attached 
[Klippy.log](https://github.com/KevinOConnor/klipper/files/2901080/Klippy.log)

KevinOConnor, 2019-02-25:
This appears to be a defect in Klipper.

-Kevin

KevinOConnor, 2019-02-26:
This should be fixed now (commit ff9605c0).

-Kevin

closed 2019-03-07 by KevinOConnor