Klipper3d/klipper #1289: Move exceeds maximum extrusion error when adjusting Z Offset

d-lee, 2019-02-20:
About 3 out of 5 times when I try to adjust Z offset via the Tune menu during the print, I get "Move exceeds maximum extrusion" error. 

[klippy.log](https://github.com/KevinOConnor/klipper/files/2885311/klippy.log)

KevinOConnor, 2019-02-22:
There is a problem in Klipper if one issues a SET_GCODE_OFFSET command immediately prior to an extrude only move.  Unfortunately, until this can be addressed, the only workaround is to limit the use of SET_GCODE_OFFSET.

-Kevin

d-lee, 2019-03-02:
Okay. That makes sense. I suspected that something along these lines was happening. Should I keep the issue open or is it a duplicate of another known issue?

KevinOConnor, 2019-03-04:
I'd keep it open.

-Kevin

DruckiMcDruck, 2019-04-10:
I also have this issue. My current workaround is to just issue the Z offset via Octoprint terminal before starting the print. You'll have to know the offset though... As my offset usually doesn't change for weeks, i just apply it like this.

KevinOConnor, 2019-05-21:
With the change in PR #1662, I believe this issue will be fixed.

-Kevin

KevinOConnor, 2019-06-02:
PR #1662 (commit d6b69938) was merged.

-Kevin

closed 2019-06-02 by KevinOConnor