Klipper3d/klipper #1292: Some files are not updating in github

mtsgsd, 2019-02-21:
Hard to condense to title. The BLtouch.md file in https://github.com/KevinOConnor/klipper/tree/master/docs shows as being 26 days old. Meanwhile that same document in the downloaded version is very new (and excellent btw, big improvement).

I'm wondering if there is more that may not be updating online. Thought you would want to know. People can't refer to the online files and get the latest version without downloading them.  I did clear my Chrome cache, and it made no difference.

cosminr86, 2019-02-21:
What I think is happening is this:
The date you see on github is the date the file was created. 
On a new feature there is a fork of master branch. That files can take a long time from creation to the final file that is merged in master branch.

mtsgsd, 2019-02-21:
it's not just the date.  compare the contents. Shep confirmed this for me. So I'm not crazy. :)

KevinOConnor, 2019-02-22:
Not sure what occurred - I see the latest version.  Maybe some github snafu.

-Kevin

closed 2019-03-04 by KevinOConnor