Klipper3d/klipper #1294: [NO ISSUE] Convert M204 to SET_VELOCITY_LIMIT

Hywelmartin, 2019-02-21:
Do as you like with it...
#362 #1280 #282 
```
#!/usr/bin/perl -i         

# this script is intended to use in Slic3r PE to be able
# to change accelerations in conjunction with retardation
# in Klipper
# make sure to make the file executable with chmod +x

# the script  looks for:
# M204 Sxxxx
# and convert them to:
# SET_VELOCITY_LIMIT ACCEL=xxxx ACCEL_TO_DECEL=(xxxx*0.5)

# The script can also be used as a stand alone script
# $ cat input.gcode | ./post.pl > out.gcode
                                               
use strict;
use warnings;

while (<>) {

    if (/^M204(\h+S(-?\d*\.?\d+))?/) {
	my $S=$2; 
        my $R=$S*0.5;
        $_ = "SET_VELOCITY_LIMIT ACCEL=$S ACCEL_TO_DECEL=$R\n";
    }

    print;
}
```

FHeilmann, 2019-02-22:
can't you use a gcode macro with parameters for this?

rmcbc, 2019-02-22:
I think that you can use something like this:

[gcode_macro M204]
default_parameter_S=1000
#   Default 1000 if S is omitted
gcode:  SET_VELOCITY_LIMIT ACCEL={S} ACCEL_TO_DECEL=({S}*0.5)

Hywelmartin, 2019-02-22:
https://github.com/KevinOConnor/klipper/issues/1280

FHeilmann, 2019-02-22:
@rmcbc unfortunately the math expression will not work. I thought this could be done with a macro as well but since we need to halve accel to decel it won't work this way

Hywelmartin, 2019-02-22:
And hijacking an existing gcode don't work either

KevinOConnor, 2019-03-11:
Thanks.  If there's no further comments on this, I'm inclined to close this issue.

-Kevin

Hywelmartin, 2019-03-11:
@kevinoconnor is there a place for tips and tricks..

KevinOConnor, 2019-03-11:
Good question.  I don't really know of a good place.  I'm open to suggestions.  I don't think github issues are a good (long term) location.

-Kevin

JohnEdwa, 2019-03-12:
A Wiki?  
[That's how Marlin does it](https://github.com/MarlinFirmware/Marlin/wiki), and I really can't think of any other format that would work if the idea is that it's maintained by the community and not by you directly (which you currently do, it's just called [the FAQ](https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md)).

theopensourcerer, 2019-03-12:
Seen this (Someone linked to it from the Klipper group on Facebook)? It looks like it takes the docs direct from Github but presents them in a easier to read fashion...

https://drphil3d.github.io/klipper/#/

Perhaps this kind of thing could be used to pull in docs from more than one source? I'm not familiar with github io myself so just thought I'd throw it in.

closed 2019-04-02 by Hywelmartin