Klipper3d/klipper #1295: GET_PIN

thucar, 2019-02-22:
Is there any way to return the known state of a pin with Klipper? I'm using ps_on pin to turn my ATX power supply on and off. Since I could not find a way to handle the switching directly from Klipper, I've delegated the task to Octoprints PSU Control plugin. The only thing I'm missing right now is getting accurate feedback about the PSU status back to Octoprint. 

I figured if I could return the known state of ps_on pin with a gcode macro, then that would solve my problem. Unfortunately I have not found any means to retrieve that state in Klipper. Is it something that is not possible as is and would need a code change?

AxMod3DPrint, 2019-02-22:
For RAMPS:

```
[output_pin ps_on]
pin: !ar12
value: 0
shutdown_value: 0

[gcode_macro m80]
gcode:
  SET_PIN PIN=ps_on VALUE=1

[gcode_macro m81]
gcode:
  SET_PIN PIN=ps_on VALUE=0
```

thucar, 2019-02-22:
Thanks but that part I've got sorted :)

The problem is, right now I'm limited to using "Internal sensing" on the PSU control plugin. But I would like to use "System command" to get actual feedback from Klipper.

AxMod3DPrint, 2019-02-22:
System command relates to the GPIO of the Pi, which I see where you're coming from, but I believe would require development.

KevinOConnor, 2019-02-22:
It is not currently possible to query the state of an arbitrary pin via g-code.  It is possible to do that from a python "extras" module.  See docs/Code_Overview.md if interested in doing development.

-Kevin

closed 2019-02-23 by thucar

fessyfoo, 2019-03-24:
link for the lazy [docs/Code_Overview.md](https://github.com/KevinOConnor/klipper/blob/master/docs/Code_Overview.md)