Klipper3d/klipper #1296: Can't connect after update

Avalonnw, 2019-02-22:
After updating perfectly working system (raspbian on a laptop) via apt-get and updating klipper I can't connect to printer anymore:
`Connecting to: /tmp/printerChanging monitoring state from "Offline" to "Error: Connection error, see Terminal tab"Unexpected error while connecting to serial port: /tmp/printer SerialException: '[Errno 13] could not open port /tmp/printer: [Errno 13] Permission denied: '/tmp/printer'' @ comm.py:_openSerial:2478 (hook default)`

Might be related, but I get pip version 10 if I type pip --version but updating it with pip install --upgrade pip reports version 19.0.1. Can't solve this at all.

AxMod3DPrint, 2019-02-23:
Please attach your klippy.log. 

In the mean time, some simple verifications

Check `/tmp/printer` exists -
Login via ssh
`ls -al  /tmp/`

Check Klipper is running -
```
sudo service klipper status
```
Output if running - 
```
● klipper.service - LSB: Klipper daemon
   Loaded: loaded (/etc/init.d/klipper; generated; vendor preset: enabled)
   Active: active (running) since Tue 2019-02-19 10:17:05 UTC; 3 days ago
     Docs: man:systemd-sysv-generator(8)
  Process: 257 ExecStart=/etc/init.d/klipper start (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/klipper.service
           └─333 /home/pi/klippy-env/bin/python /home/pi/klipper/klippy/klippy.py /home/pi/printer.cfg -l /tmp/klippy.log

Feb 19 10:17:04 hathor-pi systemd[1]: Starting LSB: Klipper daemon...
Feb 19 10:17:05 hathor-pi klipper[257]: Starting klipper: klipper.
Feb 19 10:17:05 hathor-pi systemd[1]: Started LSB: Klipper daemon.
```
Look for the Active line - It will either say Active (running), Active (exited) or Stopped.  the latter two mean the klipper service isn't running.  Stopped is just that it hasn't started, Active (exited) means it's crashed and we will need the log to figure out why.

Avalonnw, 2019-02-23:
Ok. Testing:
`ls -al /tmp/
total 48
drwxrwxrwt 10 root root 4096 Feb 23 18:17 .
drwxr-xr-x 22 root root 4096 Jun 26  2018 ..
drwx------  2 pi   pi   4096 Feb 22 21:37 dhcpcd-pi
drwxrwxrwt  2 root root 4096 Feb 22 21:35 .font-unix
drwxrwxrwt  2 root root 4096 Feb 22 21:35 .ICE-unix
-rw-r--r--  1 root root    8 Feb 23 18:22 printer
drwx------  2 pi   pi   4096 Feb 22 21:36 ssh-dmItKFnMUHXM
drwx------  2 pi   pi   4096 Feb 22 21:36 ssh-zYKVZQeKlE1L
drwxrwxrwt  2 root root 4096 Feb 22 21:35 .Test-unix
-r--r--r--  1 root root   11 Feb 22 21:36 .X0-lock
drwxrwxrwt  2 root root 4096 Feb 22 21:36 .X11-unix
drwxrwxrwt  2 root root 4096 Feb 22 21:35 .XIM-unix
`
` $ sudo service klipper status
● klipper.service - LSB: Klipper daemon
   Loaded: loaded (/etc/init.d/klipper; generated; vendor preset: enabled)
   Active: active (exited) since Fri 2019-02-22 21:52:14 ACDT; 20h ago
     Docs: man:systemd-sysv-generator(8)
  Process: 2491 ExecStop=/etc/init.d/klipper stop (code=exited, status=0/SUCCESS
  Process: 2503 ExecStart=/etc/init.d/klipper start (code=exited, status=0/SUCCE
    Tasks: 0 (limit: 4915)
   CGroup: /system.slice/klipper.service

Feb 22 21:52:14 3dprinter systemd[1]: Starting LSB: Klipper daemon...
Feb 22 21:52:14 3dprinter klipper[2503]: Starting klipper: klipper.
Feb 22 21:52:14 3dprinter systemd[1]: Started LSB: Klipper daemon.
`

Avalonnw, 2019-02-23:
I'm probably blind, but I can't find log...

AxMod3DPrint, 2019-02-23:
No log created, but the port is there, and klipper's failed to start. Which is why it's failing to connect. It could be something as basic as the /tmp/printer port not being cleared on restart.

Try:
Rename old port:
`mv /tmp/printer /tmp/printer.old`

Restart Klipper
`sudo service klipper restart`

Avalonnw, 2019-02-23:
Ok, that sorta worked:
`Recv: ok FIRMWARE_VERSION:v0.7.0-209-gda68da7 FIRMWARE_NAME:KlipperSend: M115Recv: ok FIRMWARE_VERSION:v0.7.0-209-gda68da7 FIRMWARE_NAME:Klipper[...]Send: G28Recv: // Printer is not readyRecv: // The klippy host software is attempting to connect.  PleaseRecv: // retry in a few moments.Recv: !! Printer is not readyRecv: ok[...]Send: restartRecv: okRecv: // Klipper state: Disconnect[...]Send: G28Recv: // Printer is not readyRecv: // The klippy host software is attempting to connect.  PleaseRecv: // retry in a few moments.Recv: !! Printer is not readyRecv: ok[...]Send: statusRecv: // Printer is not readyRecv: // The klippy host software is attempting to connect.  PleaseRecv: // retry in a few moments.Recv: // Klipper state: Not readyRecv: !! Printer is not readyRecv: ok`

I was able to get log. 
[klippy.log.config0001.cfg.txt](https://github.com/KevinOConnor/klipper/files/2896832/klippy.log.config0001.cfg.txt)
[klippy.log.txt](https://github.com/KevinOConnor/klipper/files/2896833/klippy.log.txt)

Avalonnw, 2019-02-23:
Aaand.... back to original issue after restarting host PC...

AxMod3DPrint, 2019-02-23:
Honestly, I'd try running the Klipper uninstall script and re-installing it first.

Avalonnw, 2019-02-25:
No luck with uninstalling. No luck with reinstalling OS. Will try Ubuntu next.

cmock, 2019-02-25:
I think you need the full path to the USB port (i.e. /dev/something/ prepended):

    [mcu]
    serial = usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0

In the log it is complaining about this with "no such file or directory"...

KevinOConnor, 2019-02-25:
As mentioned above, the `serial` setting in the config is incorrect.  It also looks like the software is running on a machine other than a Raspberry Pi.  As a guess, some startup script is doing something like `echo abc > /tmp/printer` as root, which is creating a "root owned" file that prevents Klipper from creating the pseudo-tty at /tmp/printer.

-Kevin

Avalonnw, 2019-03-01:
Ok.That is official. I am an idiot. /dev/serial/by-id/ was needed.
I don't remember deleting it, but obviously it was missing.
Thank you everybody.

closed 2019-03-01 by Avalonnw

apprentice10, 2020-07-16:
Wait! so How did you solve this again? I'm having same problem but I didn't update anything, I made 1 print and for the next one gave me this error