Klipper3d/klipper #1297: Feature request: deceleration when homing or probing

nythil, 2019-02-22:
I've been debugging the issue of unreliable probing with the dc42 IR height sensor. In the end, it turned out to be a mechanical weakness in my machine due to the springiness of a cantilever bed platform. While the issue is fully resolved by a more rigid mechanical construction, I believe it's common enough that it warrants creating a software/firmware workaround.

Upon detecting that the probe is triggered, Klipper FW immediately [stops scheduling steps](https://github.com/KevinOConnor/klipper/blob/master/src/endstop.c#L71) and [cancels all queued moves](https://github.com/KevinOConnor/klipper/blob/master/src/stepper.c#L270-L285). This action breaks the motor violently, leaving a lot of kinematic energy that needs to be dissipated. In my case it was via spring action of the bed platform. Perhaps steps are lost in the process too if the motor is unable to apply sufficient torque.

One workaround for this issue is greatly reducing probing speed. Less kinematic energy means less dissipation via bed springiness. A compromise between speed and accuracy would be to implement fast probe / backoff / slow probe, similarly to how endstops work. I intend to do that (#1249).

Another, ideal workaround would be to **implement deceleration when a probe is triggered**. This could be used by the homing procedure too in order to reduce wear on the mechanics. Obviously, deceleration rate would need to be very fast. However, even very fast deceleration would be better than the violent halt that is happening now.

KevinOConnor, 2019-02-25:
My recommendation would be to configure a lower probe speed.  I suppose one could implement the "fast probe", "slow probe" mechanism, but that seems unnecessary as one should be able to know the location of the probe to within a few millimeters and thus skip the "fast probe" step.

Performing deceleration after the trigger point would have high software complexity.

I don't think lost steps are an issue with any sane probing speed.  A stepper motor can not lose "microsteps" - it can only lose steps in increments of "4 full steps" (so, if the driver is using 16 microsteps then the stepper motor could only lose steps in increments of 64).

-Kevin

KevinOConnor, 2019-03-22:
I'm going to close this as it appears the discussion has concluded on it.

-Kevin

closed 2019-03-22 by KevinOConnor