Klipper3d/klipper #1298: Problem with GLPI

skazznj, 2019-02-22:
Hello, I'm trying to update GLPI and wamp and import the data from the old GLPI into the new one. I downloaded wap and set it up, on phpMyadmin I created a database in which I imported the old data then I try to install GLPI by updating the database (which contains the old data) however when I want to update I have this message 
Mise à jour
Connexion à la base de données réussie

La version de la base de données semble correcte (5.7.24) - Parfait !La version de la base de données semble correcte (5.7.24) - Parfait !
Mise à jour en 9.3.2

Changement de la structure de la base de données - glpi_states (0 seconde)
PHP User Warning: Table glpi_states does not exists in C:\wamp64ajours\www\glpi\inc\dbmysql.class.php at line 704
PHP User Warning: Table glpi_states does not exists in C:\wamp64ajours\www\glpi\inc\dbmysql.class.php at line 704
PHP User Warning: Table glpi_states does not exists in C:\wamp64ajours\www\glpi\inc\dbutils.class.php at line 457
PHP User Warning: Table glpi_states does not exists in C:\wamp64ajours\www\glpi\inc\dbutils.class.php at line 457
9.3.2 multiple alter in glpi_states - Erreur durant l'éxecution de la requête : ALTER TABLE `glpi_states` ADD `is_visible_enclosure` TINYINT(1) NOT NULL DEFAULT '1' AFTER `is_visible_rack` , ADD `is_visible_pdu` TINYINT(1) NOT NULL DEFAULT '1' AFTER `is_visible_enclosure` , ADD INDEX `is_visible_enclosure` (`is_visible_enclosure`) , ADD INDEX `is_visible_pdu` (`is_visible_pdu`) - L'erreur est La table 'glpi.glpi_states' n'existe pas 
Pouvez-vous m'aider s'il vous plait

manu7irl, 2019-02-22:
wrong place mate...

KevinOConnor, 2019-02-25:
This issue appears to have been opened in the wrong github repository.

closed 2019-02-25 by KevinOConnor