Klipper3d/klipper #1299: Initial button release does not trigger callback

alecplumb, 2019-02-22:
If I register a button callback, and the button is pressed/triggered when Klipper starts, then the first time the button is released the callback is not called. All further presses and releases do call the callback. If the button is released when Klipper starts, then the first press does call the callback correctly.

See: https://github.com/alecplumb/klipper/blob/work_gcode_button/klippy/extras/gcode_button.py

The same problem affects the filament_runout_sensor from PR#1098. If there is filament present when Klipper starts, then the runout sensor will not trigger.

alecplumb, 2019-02-22:
After some further testing, this only seems to be a problem if the button pin is inverted.

alecplumb, 2019-03-01:
Again more testing: This issue is present on my DuetWifi MCU, but not on RAMPs. It also affects the `[filament_switch_sensor]` extra.

KevinOConnor, 2019-03-04:
Sounds like the initial buttons.py code startup isn't propagating the base state correctly for inverted pins.  I have not had a chance to look further at this.

-Kevin

Arksine, 2019-07-05:
Just to update this issue, adding a dummy [gcode_button] with an unused pin pulled high (on the same MCU as the filament sensor or gcode button) works as a temporary workaround.

KevinOConnor, 2019-07-05:
FYI, I believe the underlying issue is that if the micro-controller buttons.c code starts with all registered pins in an off state then it doesn't report it to the host code. That's okay if all buttons are trigger high.  However, if a button is trigger low, then it means it is "triggered" at startup, but the mcu doesn't actively inform the host of that.

I suspect the best fix would be for the host code to recognize that if it doesn't get a button status message on startup, then it should recognize that any trigger low buttons have started in trigger mode.  

-Kevin

KevinOConnor, 2019-07-06:
FYI, hopefully this is now fixed as of PR #1788.

-Kevin

closed 2019-07-18 by KevinOConnor