Klipper3d/klipper #1300: Request PROBE_ACCURACY minimum Z able to be reduced to 5mm

bruce356, 2019-02-23:
Hi KevinOConnor thanks for the new feature, it works well for me but making minimum Z height 10mm it takes longer than it needs to. Default at 10mm by all means is fine but I run my BLtouch at 5mm for bedlevel etc otherwise raising the Z by 10mm each time is just too slow on doble touch/sense because Klipper does not have dual speeds (as far as I know) for Z sensing. Would be nice if you could allow a height less to 10mm to be set.

thanks - bruce

jakep82, 2019-02-23:
The Z height can be set.  It's all in the docs.

https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md#probe

bruce356, 2019-02-23:
@jakep82 thanks that was a quick reply, unless things have changed since yesterday or I have done something wrong, when I attempted to set [Z=<z height>] at less than 10mm I received an error telling me that 10mm is the minimum setting.
regards - bruce

KevinOConnor, 2019-02-25:
@rmcbc - do you have any comments on this?

-Kevin

rmcbc, 2019-02-25:
@KevinOConnor , there is no problem I can change the min to 5 (or less) and let the default as 10.
If you want I change it and create a PR.
@bruce356 what do you think?

KevinOConnor, 2019-02-25:
@rmcbc - Yes, thanks.  My recommendation would be to only enforce a `minval=self.z_offset`.

-Kevin

rmcbc, 2019-02-25:
@KevinOConnor when I get home I'll do it. Thanks for you recomendation.

rmcbc, 2019-02-25:
After validation this issue is fixed by the pull request "PROBE_ACCURACY: Set z min value to z_offset. #1314"

rmcbc, 2019-02-25:
@bruce356 the code was merge into the master so know you can use it.

bruce356, 2019-02-26:
@rmcbc thanks very much, I go as low as 2 mm on repeat probing when mesh bed levelling but for PROBE_ACCURACY testing 5mm is probably better.
Regards - bruce

closed 2019-02-26 by bruce356