Klipper3d/klipper #1302: Did I kill something or is it just my config file?

thucar, 2019-02-23:
I added ATX PSU PS_ON and a servo for a bed probe. Foolishly I did not back up my old config file. At any rate, after finishing everything, I'm able to control the PSU just fine, however the servo does not move and neither do any of my steppers.

When trying to stepper_buzz, there's nothing and when trying to home axis, it throws an error a few seconds later about endstop timeout.

I've double checked my wiring and everything seems as it should. Is there anything in my config file that sticks out to anyone as a potential culprit? I'm running RAMPS 1.4 and my servo is connected to the first servo port on RAMPS (D4) with power being pulled from the +5V pin next to PS_ON/VCC pins.

```
[stepper_x]
step_pin: ar54
dir_pin: ar55
enable_pin: !ar38
step_distance: .0125
endstop_pin: tmc2130_stepper_x:virtual_endstop
position_min: -53
position_endstop: -53
position_max: 200
homing_speed: 50

[tmc2130 stepper_x]
cs_pin: PK1
microsteps: 16
interpolate: True
run_current: 0.8
hold_current: 0.3
stealthchop_threshold: 100
driver_SGT: 12
diag1_pin:^!ar3

[stepper_y]
step_pin: ar60
dir_pin: !ar61
enable_pin: !ar56
step_distance: .0125
endstop_pin: tmc2130_stepper_y:virtual_endstop
position_min: -13
position_endstop: -13
position_max: 200
homing_speed: 60

[tmc2130 stepper_y]
cs_pin: PG1
microsteps: 16
interpolate: True
run_current: 0.8
hold_current: 0.3
stealthchop_threshold: 100
driver_SGT: 14
diag1_pin:^!ar14

[stepper_z]
step_pin: ar46
dir_pin: ar48
enable_pin: !ar62
step_distance: .00025
endstop_pin: probe:z_virtual_endstop
position_min: -10
position_max: 210

[tmc2130 stepper_z]
cs_pin: PL7
microsteps: 16
interpolate: True
run_current: 0.8
hold_current: 0.2
stealthchop_threshold: 2

[probe]
pin: ^!ar19
#   Probe detection pin. This parameter must be provided.
x_offset: 51.0
#   The distance (in mm) between the probe and the nozzle along the
#   x-axis. The default is 0.
y_offset: 0.0
#   The distance (in mm) between the probe and the nozzle along the
#   y-axis. The default is 0.
z_offset: 0.65
#   The distance (in mm) between the bed and the nozzle when the probe
#   triggers. This parameter must be provided.
speed: 1.0
#   Speed (in mm/s) of the Z axis when probing. The default is 5mm/s.
#activate_gcode:
#   A list of G-Code commands (one per line; subsequent lines
#   indented) to execute prior to each probe attempt. This may be
#   useful if the probe needs to be activated in some way. Do not
#   issue any commands here that move the toolhead (eg, G1). The
#   default is to not run any special G-Code commands on activation.
#deactivate_gcode:
#   A list of G-Code commands (one per line; subsequent lines
#   indented) to execute after each probe attempt completes. Do not
#   issue any commands here that move the toolhead. The default is to
#   not run any special G-Code commands on deactivation.

[endstop_phase]

[homing_override]
set_position_z: 0
axes: z
gcode:
	G0 F1500
	G0 Z5
	G0 X142 Y15
	G28 Z   # triggering 2 times
	G0 Z10

[extruder]
step_pin: ar26
dir_pin: !ar28
enable_pin: !ar24
#step_distance: .0021638
#step_distance: .0021123
step_distance: .00190107
nozzle_diameter: 0.25
filament_diameter: 1.750
min_extrude_temp: 170
max_extrude_only_distance: 100
max_extrude_only_velocity: 200
max_extrude_cross_section:1
pressure_advance: 0.12
pressure_advance_lookahead_time: 0.010
heater_pin: ar10
sensor_type: ATC Semitec 104GT-2
sensor_pin: analog14
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
min_temp: 0
max_temp: 300

[tmc2130 extruder]
cs_pin: PK3
microsteps: 16
interpolate: True
run_current: 0.6
hold_current: 0.2
stealthchop_threshold: 200

[heater_fan my_nozzle_fan]
pin:ar9
heater: extruder
heater_temp: 50.0
fan_speed: 1.0

[heater_bed]
heater_pin: ar8
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog13
#control: watermark
min_temp: 0
max_temp: 150

[mcu]
serial: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
pin_map: arduino

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 900
max_z_velocity: 3.1
max_z_accel: 6

[output_pin ps_on]
pin:!ar12
value: 0
shutdown_value: 0

#[pause_resume]
#recover_velocity: 50.

[gcode_macro m80]
gcode:  SET_PIN PIN=ps_on VALUE=1

[gcode_macro m81]
gcode:  SET_PIN PIN=ps_on VALUE=0

[servo touch]
pin: PG5
#maximum_servo_angle: 180
#minimum_pulse_width: 0.0006
#maximum_pulse_width: 0.0024

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [heater_bed]
#*# control = pid
#*# pid_kp = 69.219
#*# pid_ki = 1.775
#*# pid_kd = 674.884
#*#

```

AxMod3DPrint, 2019-02-23:
Pin Assignments - looks like you're on a RAMPS, unless you have a second mcu pins should start ar

thucar, 2019-02-23:
I found it!

The pin assignments are ok. The ar# are just aliases.

However, when making the changes, I also cleaned out most of the comments that I had copied from the examples. Oddly enough, the drivers section in TMC2130 is required, even though it's commented out.

After adding the block below to each TMC2130 config section, it's all working again!

```
#driver_IHOLDDELAY: 8
#driver_TPOWERDOWN: 0
#driver_BLANK_TIME_SELECT: 1
#driver_TOFF: 4
#driver_HEND: 7
#driver_HSTRT: 0
#driver_PWM_AUTOSCALE: True
#driver_PWM_FREQ: 1
#driver_PWM_GRAD: 4
#driver_PWM_AMPL: 128

```

trevjonez, 2019-02-23:
@thucar that seems like a bug or not actually what your problem was. best thing to do is the review the klippy log it will have a stack dump of what the actual issue is.

CHILLYSMOKES, 2019-02-23:
TMC2130s +PS_ON again. 

Please see issue #1277. Read the whole thing, bro.

thucar, 2019-02-24:
Thanks @CHILLYSMOKES That explains a lot.

closed 2019-02-24 by thucar