Klipper3d/klipper #1303: Virtual SD: Directory Support

trevjonez, 2019-02-23:
I have recently switched to using the virtual SD for everything and gcode organization has become tedious at best. I started moving things into directories to better organize things but then I realized klippy assumes a single flat dir for the virtual sd. 

I have not looked any deeper yet as I am busy with other tasks, just wanted to get the feature request tracked.

KevinOConnor, 2019-02-25:
To the best of my knowledge, directories are not generally supported via the g-code "sd card printing" mechanism.  Thus, I suspect this isn't viable.

-Kevin

trevjonez, 2019-02-26:
looks like M20 in terms of reprap defines S & P optional args. https://reprap.org/wiki/G-code#M20:_List_SD_card

P seems to be the missing element here.

additionally the display system would need to be updated to support it as it already lists directories but can't do anything with them.

KevinOConnor, 2019-02-26:
Gcode is terrible.  From my perspective, what the gcode "documentation" states is not relevant.  The only thing that matters is being compatible with commonly used software in their standard configurations.  It was my understanding that common 3rd party firmware and common 3rd party front-ends did not support directories.  Thus, my opinion is that it does not make sense to add support to Klipper.

-Kevin

closed 2019-02-27 by trevjonez