Klipper3d/klipper #1304: Virtual SD: Consistent Sorting

trevjonez, 2019-02-23:
As a user I want virtual sd file listings to be in a consistent predictable manner.

Currently the virtual SD listing happens via `os.listdir` which when I look it up mentions `The list is in arbitrary order`. stackoverflow reponses seem to indicate it is based on the indexing of the particular filesystem.  So while it may be consistent for some there is no guarantee.

It looks like #846 touched on this briefly but it might be worth adding some configuration options for it.

Perhaps something like:
```
[virtual_sdcard]
path: ~/gcode
sorted_by: [newest | oldest | name | etc...]
```

Actual implementation at configuration time could create a lambda to pass as the key argument later?

KevinOConnor, 2019-02-25:
If someone wishes to add a sort order to the file listing and submit a github pull request, then that would be great.  I don't think it is worthwhile to configure the sort order (I think that is the role of the software displaying the listing to the user).

-Kevin

trevjonez, 2019-02-26:
giving the user the option to customize sort order seems to be a normal feature in modern software is why I bring up. 

I don't think it needs to be a runtime parameter in klipper as its configuration file setup gives a nice declarative approach to handling user needs.

A default sorted_by choice of alphanumeric on the file name would make this an invisible change to existing users.

KevinOConnor, 2019-03-11:
I'm going to close this issue.  Feel free to open a pull request when ready.

-Kevin

closed 2019-03-11 by KevinOConnor