Klipper3d/klipper #1305: M109 Overheating and then waiting for temp drop back.

pbogut, 2019-02-23:
I've installed klipper today, I'ts working now, I will be testing it in next days.
What I did notice is that my start gcode takes longer then with Marlin and then notice temp issue.

My starting gcode:

```
M107
;Sliced in Slic3r
G21        ;metric values
G90        ;absolute positioning
M82        ;set extruder to absolute mode
M107       ;start with the fan off

M140 S60 ;Start heating bed
G28 ; Home position
M190 S60 ;Wait for bed to reach temp before proceeding

G28 Z ; Re-home only Z after bed is heated

M104 S175 ; Start heating extruder

G29 ; Autolevel while hotend is heating

G0 X-2 Y05 F9000 ; Go to front

; ISSUE STARTS HERE
M109 S200 ;Wait for extruder to reach temp before proceeding

; Clean Nozzle
G0 Z0 ;drop to brush, close hall on in the process
```

After sending `M109 S200` tool is heated to 200, then I can see this line in klipper log:

```
Heater extruder within range of 200.000
```

I would expect that nozzle cleaning starts right after temp reaches 200, but instead it goes up to 210-ish then drops back below 200, and then nozzle clean code is starting (or sometimes its dropping to much and starts heating again).

My klipper version: da68da7a63322c300acf574f0068bb09b8073916

[full-klippy.log](https://github.com/KevinOConnor/klipper/files/2897143/full-klippy.log)
[klippy.log starting right before print starts](https://github.com/KevinOConnor/klipper/files/2897133/klippy.log)
[test.gcode.zip](https://github.com/KevinOConnor/klipper/files/2897135/test.zip)

JamieAKing, 2019-02-23:
I had the exact same problem thou it was over shooting + and - so much that clipper never released the wait! after much searching and hair pulling I found a very simple command to rectify it
run the following, one at a time in octoprint terminal
PID_CALIBRATE HEATER=extruder TARGET=215
PID_CALIBRATE HEATER=heater_bed TARGET=60

don't forget to use 
SAVE_CONFIG
after each one

now to post my z problem ........

pbogut, 2019-02-23:
Thanks a lot, that helped.

closed 2019-02-23 by pbogut