Klipper3d/klipper #1306: Help Please, what have I missed?

JamieAKing, 2019-02-23:
My printer build is finally complete and after a few inevitable crashes while I sorted out the homing and probing routines I tried to run the first print ..........

so far I have not successfully got past the first 20 or so layers. z stops running 
I have the dreaded tmc 2208 drivers  but have them in uart with stealth chop threshold set above maximum axis speed 
I have also followed the guide in issue #196 and fixed the pmw grad

all runs fine during homing and probing etc and can see the bed mesh levelling working during print, but
after a few layers all movement stops and things get messy

I should also say I have fade_end = 0 so as to disable that. my bed has a fair shape to it (will be getting a milled tooling plate soon thou maybe changing to 2130's first) 

I don't think that it is an over heat problem as I have the controller_fan thermistor wedged in-between the drivers and the attached 80mm controller fan directly over top of them. as you can see from the log file the temp never exceeds 34ºC (set point is 30) I have my eye on the recent commit regarding temp sensors that can be displayed in octoprints temp graphs

attached is the log with the last failed print and the tmc dump for z

I'm sure its something I have done wrong I just can't find what
any suggestions or help would be very greatly appreciated 
[klippy.log](https://github.com/KevinOConnor/klipper/files/2897182/klippy.log)

bruce356, 2019-02-23:
Hi, not sure if you have the same setup, I have two Z axis steppers wired in parallel and I noticed very early with drivers like the TMC 2100 (I know its some time ago since they came out) and the TMC2208 Tmc2224 they are not necessarily a good reliable driver to use on the Z axis (had the same results as you after some time the Z axis would just cease to lift to the next layer), I always use a different driver (other than Trinamic) for my Z axis. That way I have no problems and due to the nature of Z noise is not really an issue either.
regards - bruce

shadrincev, 2019-02-24:
Good afternoon. Drivers TMC2208 poorly on the axes with interpolation. Even after following the recommendations of this post, you may not get a good result. I advise you to do so: 
microsteps = 256 **or** 128 **or** 64
interpolate = **False** 
don't forget to adjust your step_distance for the desired axis.

AxMod3DPrint, 2019-02-24:
I would also state, that if your TMC drivers are not actively cooled, then this can also cause issues.

cosminr86, 2019-02-24:
Afaik trinamic recommendations are to use series for 2 motors on the same driver, not parallel

JamieAKing, 2019-02-24:
Thanks guys, I was hoping it wasn’t going to be the tmc2208 they are so nice and quiet. I will try adjusting the micro stepping interpolation first if that fails I do have some dvr8825 that never got used in a cnc build (went with much bigger motors and drivers) can anyone recommend another driver that has similar properties to threat of the 2208?

JamieAKing, 2019-02-24:
I have a 3 motor z but each motor has its own driver

bruce356, 2019-02-24:
If you have DRV8825 they will do the job perfectly, but if you want a driver quieter than the DRV8825 (not as quiet as the Trinamic) then the  MKS LV8729 would be an option.

JamieAKing, 2019-02-24:
Wel I’ve been at this all weekend and I give up, the z now seems to work but the x and y randomomly stop during a print for a few seconds and the restart without setting off any alarms. So I swapped them for drv8825 and is working well but the noise is horrible, I was hoping to have this run through the night but I don’t see that ever being an option with these drivers, the best I have managed is 16x microstepping on all axis, any more and I run out of bandwidth in the f6 2560 chip. Even now large z moves lock the system. I’m assuming Klipper is trying to send the pulses for the entire move in one hit so small (100mm) moves complete fine but a 200mm move stops it dead.
The solution to z seems to have been a combination of things, reduce run and hold current by 0.1amp disable step interpolation and increase micro steps from 4x to 16x. 
None of these helped with the x and y . When z stopped they did not restart and was always all 3 motors together where as x and y just randomly pause for a few seconds. 
Before some reply’s with overheating I have heat sinks and an 80mm fan directly over the top of the board. The motor after 10mins running  are barely warm to the touch  

If anyone has any insights I would love to here them, I may even go through the ordeal of putting the tmcs back in if some one has a definitive answer for me ....

Artem-B, 2019-02-24:
First question -- are you driving the motors with 12V? If you do, it will severely limit the wiggle room you may have in tuning the drivers. Driving the motors with 24V appears to be much more reliable with TMC drivers.

In general, I guess you should start with checking what do the drivers report right after the problem happens.

Once you reproduce the problem, dump driver registers. Note that you need to do it *before* the motors are disabled. Send `dump_tmc stepper=stepper_x` or `stepper=stepper_x` and check the values of `GSTAT` and `DRV_STATUS`. If `GSTAT` reports that `drv_err` is set, that would confirm that the driver did indeed experience an error. Then check the values reported by `DRV_STATUS`. It may give you a clue what's the reason for the error. Various bits are described in the [TMC220x/222x datasheet](https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC220x_TMC222x_Datasheet.pdf). Basically it could be an overtemperature condition or an open/short coil circuit detection. I think the latter is what gets false positives in case of fast-changing direction.

If it's overheating -- provide more cooling. 

If it claims open/short circuit error, make sure your physical connections to the motor are solid (check that the end-to-end impedance measured with ohmmeter does not change while you're wiggling the wires/connectors between the driver and the motor). 

If physical connectivity looks OK, check for mechanical issues -- move the axis and check for bumps/unevenness and make sure the movement is smooth. 

Only then start tinkering with the driver settings. The optimum will be somewhere between the current low enough that it does not provide enough torque to accelerate/decelerate and the nominal coil current. It's also affected by whether you drive with 12V or 24V. 12V may make it impossible to reach desired current on the given motor fast enough and that will confuse the drivers. 

**NOTE**: Take all that I've said above it with a big grain of salt as I don't have specific details about how exactly the drivers detect the errors, so I'm speculating based on the few cases I've observed myself and what's been reported by klipper and marlin users.

shadrincev, 2019-02-25:
I'd like to ask you for a new klippy.log. And so I can only assume that you put the 256 microsteps for each driver, although he need only z axis. If you put all the drivers 256, and disable interpolation, it will be an error and the printer will stop due to too heavy calculations. From my observations, I can say that the extruder should be no more than 64 steps, for the x, y axes no more than 128, and for z you select in the ratio of silence and stability, but this is when interpolation is disabled. I would, in your place, turned off the interpolation of only the z-axis.

JamieAKing, 2019-02-25:
> First question -- are you driving the motors with 12V? If you do, it will severely limit the wiggle room you may have in tuning the drivers. Driving the motors with 24V appears to be much more reliable with TMC drivers.

Yes I am using 24v for all motors,

I tried the dump_tmc a few times but I’m afraid to confess I failed misserably at converting the output into anything meaningful. I did compare the result to the data sheet but found no correlation between my numbers and the fault codes. I did convert from hex to decimal on the assumption that I was working with 2 4digit hex numbers as with pwm_auto but that is likely the source of my failure
I have never encountered such things in my life as a carpenter so a little outside of my knowledge base.

I’m not near the printer atm but will post a fresh log when I get home. All the drivers are set at x16 micro steps. The z axis which is running Tmc_2208 has interpolation off. The drv8825 are only capable of 32x but setting this causes ‘mcu Timer in the past’ on any movement of either extruder or XY 
I’m happy with the 16x getting lovely print quality just hate the noise from the 8825,

Artem-B, 2019-02-25:
Recent klipper conveniently decodes register fields for you, so it should make the task a bit easier.

On a side note, it would be great if klipper could periodically check the driver for errors and report them, if any are found.

JamieAKing, 2019-02-25:
Latest log as promised. not that the Tmc drivers are commented out as I'm currently using drv8825 in place of the 2208 
again worth mentioning no errors when using this set up , I just real really want to make the 2208 work

if the 2130's are equally quiet I would have no issue in purchasing them instead

one other query I have unrelated to klipper and I completely understand if this gets shut down, 
but I have 1.6 amp motors yet setting anything above 1.2  amp on the tmcs resulted in the motor locking solid and not moving and .7 amp is skips like a nutter. is there something fundamental that I missed.
like maybe setting the pot on the driver in a uart position. I was under the impression that there would be much more range in the current setting for the motor
[klippy.log](https://github.com/KevinOConnor/klipper/files/2901624/klippy.log)

jakep82, 2019-02-25:
TMC stepstick drivers will usually overheat at 1.2 amps.  The chips themselves are rated for more power, but the small stepstick boards don't have enough copper to sink that much heat.  With heatsinks and a fan blowing on the drivers, you should be able to use 1 amp reliably, but above that you start to risk overheating.  I personally only need 750mA to run all of my steppers (I have 2130s), and I can move my printer at 300mm/s without losing steps.

JamieAKing, 2019-02-25:
oh speed isn't a problem, I can fly around at 500mm/s and not loose positional accuracy but I eat up my available 100k steps/second very quickly so I wound that back to 300 as you have which is plenty fast enough in a 300 x 300 printer. 
I have both heatsinks and an 80mm fan at full chat directly over the top of them I can't see temp being an issue

Artem-B, 2019-02-25:
Still, you should check the driver registers to confirm that it is indeed not the issue.

TMC chips have thermal pads on the bottom and require both proper board design and correct soldering in order to effectively conduct the heat away. It's possible that the board design or bad solder joint on the pad would result in insufficient chip cooling -- placing heatsink on the board or on top of the chip will not do much good in this scenario.

JamieAKing, 2019-02-26:
Ok I have a free day today, I will reinstall the tmc drives, 
So I’m know that Im definitely doing things in the right order
I will start a print when it fails I pause the print, 
issue the dump_tmc command, 
Issue the m112 
Pull the Klipper.log 
The restart the printer 
Is this correct?

KevinOConnor, 2019-03-07:
Yes.  I'd return to stock settings.  Set the stealthchop_threshsold to 10% above the maximum velocity for the axis (note with corexy, the maximum velocity of the xy steppers are 1.415 times the maximum xy velocity) and then run a print.  If you get a stepper lockup during a print, then run DUMP_TMC for each of the steppers, then run M112, then post the klipper log file.

If you're seeing "print pauses", that's a separate issue.  Likely OctoPrint slow downs that would be solved via a virtual_sdcard config.

-Kevin

KevinOConnor, 2019-03-22:
I'm going to close this as it appears the discussion has concluded.

-Kevin

closed 2019-03-22 by KevinOConnor