Klipper3d/klipper #1307: Feature Request: Use Docsify GitPages for documentation

drphil3d, 2019-02-23:
I just want to ask if you would be willing to install docsify into your repo so that the documentation is easier to find, searchable and more accessible. I know you didn't want to use a Wiki because it can be edited by the public (although I just learned you can restrict access). I believe Docsify / Git Pages would give your project a nice front facing site so that people can more easily understand the project.

It would also reduce the amount of issues that are opened saving you time from having to answer the same questions.

https://docsify.js.org

Thanks Kevin for all the hard work.

drphil3d, 2019-03-10:
@KevinOConnor I just want to show you what I've been working on to make the documentation more accessible and easier to read through. 
The links on the sidebar are all generated automatically from the document.

https://drphil3d.github.io/klipper/

KevinOConnor, 2019-03-11:
Thanks.  In general, I think it is a good idea.

I did try navigating the site above and I did notice a bunch of quirks (search doesn't work, links outside the docs/ directory don't work, when clicking between links the scroll bar doesn't reset, the header bar clashes with the sidebar).  I'm not sure if this is something with my browser (firefox), just integration details, or if doscify.js.org is not mature enough.

-Kevin

drphil3d, 2019-03-17:
There are a few things to be fixed, I just didn't want to invest a ton of time into it without you seeing it incase you didn't want to use it. If you like it I certainly will make it all pretty. I definitely like it, the sidebar really helps navigate through the different sections and I like how the Gcode commands are automatically highlighted so you don't have to search through every line of the doc to find one thing.

drphil3d, 2019-03-21:
Docsify is completely automated, all that has to be done is create a menu
structure.
Then github pages automatically deploys it.

It’s literally a couple lines of short codes and two files.
-- 

Philip Mally
3D Printing & Rapid Prototyping Services
530-205-3210
pmally@gmail.com

KevinOConnor, 2019-03-22:
I'm on board.

-Kevin

KevinOConnor, 2019-04-21:
FYI, I setup a similar system using "github pages" and Jekyll.  It automatically converts the github documentation to html pages.  The results are at: https://www.klipper3d.org/

-Kevin

drphil3d, 2019-04-22:
I know, Docsify runs on github pages but has more options and a far better
user interface.

KevinOConnor, 2019-04-28:
Okay, thanks for the feedback.  FWIW, I did look through Docsify before settling on a more conventional github pages approach.  Ultimately I felt Docsify added significant complexity that I wasn't fully comfortable with.  I also felt that there is a gain to performing the markdown to html conversion on the server instead of via javascript - in particular with search engine optimization and with users that restrict javascript on their browsers.

Cheers,
-Kevin

KevinOConnor, 2019-05-16:
I'm going to close this as I think the new web page implements much of what we want.  If there are customizations to the web page that make sense then feel free to open a new issue or submit a PR.

-Kevin

closed 2019-05-16 by KevinOConnor