Klipper3d/klipper #1308: unable to seek file (vsd)

maxcom99, 2019-02-24:
klipper log:
010 rttvar=0.004 rto=0.027 ready_bytes=0 stalled_bytes=0 freq=15999986 heater_bed: target=0 temp=32.2 pwm=0.000  print_time=3.285 buffer_time=0.000 print_stall=0 extruder: target=0 temp=30.9 pwm=0.000
Starting SD card print (position 0)
virtual_sdcard seek
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/extras/virtual_sdcard.py", line 138, in work_handler
    self.current_file.seek(self.file_position)
AttributeError: 'NoneType' object has no attribute 'seek'
Unable to seek file

running latest master reinstalled everything, no dice

KevinOConnor, 2019-02-25:
In order for us to assist you must attach the Klipper log file from the event - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

maxcom99, 2019-02-27:
@KevinOConnor I included the log at the top or are there more relevant sections that I missed and you want to see the whole thing?

KevinOConnor, 2019-03-07:
I'm closing this pending the full log file attached to this issue as described above.

-Kevin

closed 2019-03-07 by KevinOConnor