Klipper3d/klipper #1309: Klipper problem on Bigtreetech SKR V1.1

patoune08, 2019-02-24:
Hello,
I have a Bigtreetech SKR V1.1 board.
I installed octoprint and klipper but I can not connect octoprint to the printer.
Here is the octoprint terminal file.
Thanks for your help
[octo.log](https://github.com/KevinOConnor/klipper/files/2898175/octo.log)

patoune08, 2019-02-24:
Here is a serial connection monitoring file.
[serial.log](https://github.com/KevinOConnor/klipper/files/2898190/serial.log)

BlackStump, 2019-02-25:
The docs hold a wealth of info and for support you need to read this and supply the log mentioned.
https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

patoune08, 2019-02-25:
Hello
I had read, but the problem is that I can not issue the command M112 since I can not connect to the printer. that's why I sent you these two files.
I would just like to know how to connect with this board.
Thank you

BlackStump, 2019-02-25:
Have you read and followed the make flash info at the top of config here ?
https://github.com/KevinOConnor/klipper/blob/master/config/generic-bigtreetech-skr-v1.1.cfg

Ok I can understand you can not issue a M112 command from Octoprint but have you looked for the log file? /tmp/klippy.log  as per the instructions?

FHeilmann, 2019-02-25:
you need to connect to /tmp/printer with octoprint, not to the MCU serial port directly.

Skorpi08, 2019-02-25:
Does it only work via USB or UART?

FHeilmann, 2019-02-25:
according to the sources you can configure UART0 instead of USB

Skorpi08, 2019-02-25:
I have try it, TX to RX and RX to TX. without success

FHeilmann, 2019-02-25:
you need to configure it during the `make menuconfig` step. Disable USB serial, then the code will switch over to the uart.

patoune08, 2019-02-25:
A last post to say thank you, for now it seems to work but I think my board is dead.
I have high temperature indications on all three probes (between 170 and 180 ° C) while everything is cold.
Thanks again.

closed 2019-03-04 by KevinOConnor