Klipper3d/klipper #1311: TMC2130 sensorless homing and Endstop still triggered errors

amblidex, 2019-02-24:
I've noticed that having a non-zero `homing_retract_dist` setting on an axis (such as the default 5mm) makes homing susceptible to still triggered errors like: "Endstop x still triggered after retract". I'm guessing that `diag1` is not getting detected back to clear quickly enough before the retract movement is initiated. Marlin recommends that it's equivalent home bump setting be set to zero for sensorless homing, possibly for this reason?

When `homing_retract_dist` is set to zero, the same SGT value of 8 that works in Marlin works with Klipper. Otherwise, I've had to use a value of 20+ to avoid the error.

I don't know if Klipper should be doing anything different for retracting after sensorless homing, such as having a slight delay after triggering before allowing movement, or if this is just something to document.

timauton, 2019-02-24:
Same issue here - happy I found this report as it contains the workaround!

I have previously intermittently seen the same issue on my X-axis (which has a microswitch) too, but have been unable to reproduce that just now. It certainly feels like Klipper is expecting instantaneous response when some end stops may take a few milliseconds to spring back etc.

klippy.log attached which shows the issue with a TMC2130 on my Y-axis.

[klippy.log](https://github.com/KevinOConnor/klipper/files/2898508/klippy.log)

thucar, 2019-02-25:
I've been combating this issue as well. The odd thing is, whenever I would start getting the error, if I were to just keep trying to home axis, the error would not stop appearing even after a number of retries. But if I were to do a firmware_restart and home my axis again, the error would not show up.

Will try to set `homing_retract_dist` to zero.

KevinOConnor, 2019-02-25:
FYI, this was discussed a bit, starting at: https://github.com/KevinOConnor/klipper/pull/1004#issuecomment-447879635

FWIW, I don't own the tmc2130s myself.

-Kevin

jakep82, 2019-02-25:
I've only had success with sensorless homing using a retract distance of zero.  Any other setting always triggers an error.  This is perfectly fine in my opinion, but unfortunately this information didn't make it into the sensorless homing document.  I think the solution here is to simply document this as a requirement for sensorless homing.

amblidex, 2019-02-25:
If documenting ends up being the solution, probably should also advise users to set `position_min: -1` and `position_endstop: -1` as well, so that things aren't bumping against the limits at (0,0)

amblidex, 2019-02-26:
I think I see what's going on. When retract is enabled, I'm seeing the initial home successfully trigger, the retract move execute, then the "still triggered" error. Listening to the steppers I noticed that they weren't switching back to the quiet stealthchop mode for the retract move after the first home, and instead were in the spreadcycle mode needed for stall detection for the entire operation.

The tmc2130 virtual endstop is supposed to configure for stall detection before each home move, then revert back to regular operation afterwards. However, the SPI commands to turn off stall detection and presumably reset the diag1 indicator after the first home don't seem to be completing before the retract or second home starts - thus the already triggered error. As a test, I sprinkled in a few toolhead.wait_moves() in the homing code between moves, and this allowed the retract to operate in the quiet stealthchop mode, and the second home to execute without the still triggered error.

closed 2019-03-19 by amblidex