Klipper3d/klipper #1312: Klipper now divorced from Octoprint

BlackStump, 2019-02-25:
Primer
OctoPrint 
As a frontend is nice enough but has some issues with some prints with stuttering and wanting more resources then whats available.
example file to try yourself 
[Spline Hell.zip](https://github.com/KevinOConnor/klipper/files/2899325/Spline.Hell.zip)
Print as a single perimeter.
If I print the above file it max's the cpu on the Beaglebone Black and stutters from start to finish and yes it will print fine if I use the virtual Sdcard.

Duet Web Control for Klipper
https://github.com/Stephan3/dwc2-for-klipper
I print the same file cpu peaks at 73% but mostly low 60% and no stuttering
The GUI well for me whats not to like, fast to load and fast to operate and every thing I need and simple easy instructions to install.
I stumbled across this and I hope the author is ok me posting about his API , I am just happy there is a creditable  alternative  to Octoprint.

Maglin, 2019-02-25:
@BlackStump this is something that isn't ready for release.  The problems with free GitHubs.  This is being written by Stephan and still in testing.  @KevinOConnor this should be closed as this isn't ready to be used by the general public.  More testing is still required.  @BlackStump please refrain from bringing this up anywhere else.  Keep to it's GitHub to allow continuity please.  Loads of questions and issues will only stall development.  Thank you.

Maglin, 2019-02-25:
This can be closed.  It's not an issue.

bruce356, 2019-02-26:
I only have a RPi 3 and tested your Spline Hell.zip by saving the gcode file to a folder on the RPi did not use   virtual sd and it printed perfectly at  twice normal speed without any stuttering.
regards

BlackStump, 2019-02-26:
Hi Bruce when you say only a Rpi 3 that is a quad core cpu compared to a Beaglebone Black single core so no surprise there.
I did refer to a Beaglebone Black try it on a Rpi 1 and maybe your results will be different but my point is to the people that are having stuttering problems not to the people who are not.
Actually the real point is to mention a alternative to Octoprint and dont get me wrong if people are happy with Octoprint by all means ignore my post.
I have a Duet2 so I am familiar with the Duet Web Control and now I have a choice to run that with my Klipper firmware printers.
Choices all about choices for me I choose DWC everyone's mileage will be different but isn't that the great thing about choices?
Jumps off the soapbox!

closed 2019-02-26 by BlackStump